realsenseai/librealsense #7909: Invalid Value in rs2_set_option(options:000001BABE686D40, option:Visual Preset, value:1): The Default preset signifies that the controls have not been changed since initialization, the API does not support changing back to this state, Please choose one of the other presets

zbs881314, 2020-12-03:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0210876|
|**Firmware Version**|01.05.02.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003abfb8627|
|**Firmware Update Id**|0003abfb8627|

Please provide a description of the problem

RealSenseSupport, 2020-12-14:
Hi @zbs881314 

Can you help me understand what specifically that you do in order to get this invalid value?  If you can outline steps in which you've performed that would be most helpful.

Thanks

RealSenseSupport, 2021-01-13:
Hi @zbs881314

Can you help me understand what specifically that you do in order to get this invalid value? If you can outline steps in which you've performed that would be most helpful.

Thanks

*If we don’t hear from you in 7 days, this issue will be closed.

FSund, 2021-01-15:
I can reproduce the error the following way
 - Start Realsense Viewer v.2.40.0
 - Start L500 Depth Sensor
 - Change from Default to any of the other presets
 - Change back to Default -> error

Complete error message
```
Invalid Value in rs2_set_option(options:0000020A39B19D40, option:Visual Preset, value:1):
The Default preset signifies that the controls have not been changed since initialization, the API does not support changing back to this state, Please choose one of the other presets
```

RealSenseSupport, 2021-01-27:
Hi @FSund 

Thanks for the steps, now I understand.  This is expected and shouldn't impact functionality as moving to a preset like No Ambient Light, Low Ambient, Max Range and Short Range works properly.

closed 2021-02-26 by RealSenseSupport