realsenseai/librealsense #7912: Access one pipeline from different running scripts

alex-stonebridge, 2020-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Operating System & Version |   Ubuntu 16 | 
| Platform                                 | PC  |
| Language                            |  python                          | 
| Segment			| Robot                         | 

I have one camera and I want to access my RGB and Depth camera frames from two running applications. I thought it would be possible to start the pipeline in one script and then access it from another one, but in does not seems to be possible. Is there any approach to do that?

MartyG-RealSense, 2020-12-03:
Hi @alex-stonebridge   The **Multi-Streaming Model** section of the documentation describes the RealSense SDK's rules for accessing one camera from two separate applications (described as a **process**).  

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

Under the rule-set, only one process can access a particular stream type (**endpoint**) on a particular camera at the same time. 
 So if Process A has claimed the Depth of the single camera then Process B **cannot** also claim Depth on that same camera. 

If Process A claimed the Depth and Process B claimed the RGB though then there should not be a conflict, as Depth and RGB are different endpoints.

Once an endpoint has been 'opened' by a process, it can be 'closed', releasing ownership on the device.  I wonder if that would provide the two-applcation approach that you need.

- Open the endpoint with the first application and start the stream.
- Close the endpoint, making it available for the second application to claim.

@dorodnic   Would closing the endpoint in Process A stop the stream, or would it keep going so that Process B could claim the endpoint with an 'open' of its own?

anhTuan0712, 2020-12-04:
Hi @alex-stonebridge ,Sorry for an unrelated question, how do I do this.
[image]

alex-stonebridge, 2020-12-04:
Hi @MartyG-RealSense thanks for your point, that was really helpful. 
Also, I was thinking of using a realsense ROS wrapper and just subscribe to desired topics from different scripts. However, I have limited computational power and I am afraid that ROS approach will take much more resources compared to your approach. Is it true?

MartyG-RealSense, 2020-12-04:
@alex-stonebridge  Each situation is unique and so I would not like to make a blanket statement about performance.  You are welcome to try the ROS wrapper and see how it performs for you.  I would recommend going into it with the expectation that the results may not be exactly the same as the ones that you experience in librealsense.  The RealSense ROS wrapper may handle some functions differently from librealsense due to its adherence to ROS standards to ensure compatibility.

closed 2020-12-05 by alex-stonebridge