realsenseai/librealsense #7913: [D435i] Computing user speed from RS2_STREAM_ACCEL stream

michael-martinez, 2020-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    | 4.15.0-1101-oem                                        | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |   legacy / 2                           | 
| Language                            |  C++                          | 
| Segment			|  Computer vision                          | 

### Issue Description
I am unable to compute gross user speed estimate from D435i acceleration values by using discrete integration.

```
constexpr double SPEED_ESTIMATE_UPDATE_DELAY = 10.0;
constexpr float g = 9.81f;

 void SpeedEstimator::processAccel(rs2_vector accel_data, double ts)
 {
    double dt_accelero = (ts - last_ts_accelero) / 1000.0f;
    std::cout << "SPEED dt_accelero: " << dt_accelero << std::endl;
    
    if (last_ts_accelero == 0) {
        last_ts_accelero = ts;
        start_time = std::chrono::system_clock::now();
	    return;
    } else {
        last_ts_accelero = ts;
    }

    std::chrono::duration<double> diff = std::chrono::system_clock::now() - start_time;

    if (diff.count() < SPEED_ESTIMATE_UPDATE_DELAY) {
        accelerations.push_back(rs2_vector({ accel_data.x - g, accel_data.y - g, accel_data.z - g }));
    } else {
        auto dt = (float)dt_accelero;
        vx = 0;
        vy = 0;
        vz = 0;
        for (const auto& value: accelerations)
        {
            vx += value.x * dt;
            vy += value.y * dt;
            vz += value.z * dt;
        }
        std::cout << "SPEED Computing speed..." << vx << ", " << vy << ", " << vz << std::endl;
        
        rs2_vector speed_data = { vx, vy, vz };
       
        float l_speed = sqrt(
            pow(speed_data.x, 2.0f)
            + pow(speed_data.y, 2.0f)
            + pow(speed_data.z, 2.0f));

        std::lock_guard<std::mutex> lock(speed_mtx);
        speed = l_speed;

        start_time = std::chrono::system_clock::now();
        accelerations.clear();
    }
}

float SpeedEstimator::get_speed()
{
    std::lock_guard<std::mutex> lock(speed_mtx);
    return speed;
}
```

I am getting really high values for computed speed values:
```
SPEED dt_accelero: 0.048199
SPEED dt_accelero: 0.0642637
SPEED dt_accelero: 0.0642698
SPEED dt_accelero: 0.0642617
SPEED dt_accelero: 0.048197
SPEED dt_accelero: 0.0642607
SPEED dt_accelero: 0.0642595
SPEED dt_accelero: 0.0642563
SPEED dt_accelero: 0.0481892
SPEED dt_accelero: 0.0642554
SPEED dt_accelero: 0.0642629
SPEED dt_accelero: 0.064261
SPEED dt_accelero: 0.0481978
SPEED Computing speed...-2.24547, -73.9902, -61.238
=> speed: 96.0712

Measured accelerations are close to 9 m^s-2, 0 m^s-2 and 2 m^s-2.

```

Do you have any clue on how to perform this kind of speed calculation ?

MartyG-RealSense, 2020-12-04:
Hi @michael-martinez   Could you provide more detals about your project please.  The term 'gross speed' suggests that you are trying to calculate acceleration within a certain time frame, such as meters per second.  Is this correct please?

Also, what are you measuring the acceleration of?  Is it the camera itself, or a person carrying the camera (as suggested by 'user speed')?  Thank you very much.

michael-martinez, 2020-12-04:
Hi @MartyG-RealSense thanks a lot for your support!

'gross speed' is a misleading term, I simply meant getting user speed while he moves (speed of a person carrying the camera).

My use case is:
- When the user walks, D435i being attached to his chest front orientation, I want to get its average speed something like 0.83 m/s. 
- When the user walks faster, D435i being attached to his chest front orientation, I want to get its average speed something like 1.39 m/s. 

I do not need really precise estimates, some proportional value would be ok.
From the implementation perspective, I was thinking about clearing the acceleration buffer every 10 seconds to avoid drifts but this may be wrong.

MartyG-RealSense, 2020-12-04:
Thanks very much for the further detail.  Changes in Y-accel may not be consequential because the walker is on a relatively horizontal surface beneath their feet most of the time and so Y would fluctuate by a small amount when their body moves up and down during walking.  

If value.x, value.y and value.y store the IMU accel axis values and the units of the value are m/s squared, I would think that there would be no need to multply these values by **dt** in order to get the average speed at the moment in time, as the value should increase or decrease depending on the walking pace (which determines the meters travelled per second).

Regarding pose (position and angle) of the camera, it is possible to get these on a D435i using workarounds, such as a **SolvePnP** algorithm on OpenCV or **inference** on ROS.

https://github.com/IntelRealSense/librealsense/issues/6038#issuecomment-598089815

In regard to drift, using ROS and RTABMAP with a D435i to correct drift with **relocalization** was recently discussed.

https://github.com/IntelRealSense/realsense-ros/issues/1529#issuecomment-735299914

MartyG-RealSense, 2020-12-11:
Hi @michael-martinez   Do you require further assistance with this case, please?   Thanks!

michael-martinez, 2020-12-11:
Hi, I do not think so as we discussed the kind of algorithms involved.
For reference, I also found this interesting paper which clearly states the steps involved:
https://www.nxp.com/docs/en/application-note/AN3397.pdf

I think I will try this first before looking at more complex solutions.
I was looking for a simple up and running example as this looks like a common use case.

Thank for your support @MartyG-RealSense.

closed 2020-12-24 by MartyG-RealSense