realsenseai/librealsense #7914: L515/ The same or different depth and point (z) values

anhTuan0712, 2020-12-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Win (8.1/10   | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC |
| SDK Version                            |                         | 
| Language                            |  {C/C#}                          | 
| Segment			|                           | 

### Issue Description
Hi everyone, I'm Vietnamese, my language is not good, everyone understands.
1) D415 and D435I depth and point (z) are the same.
2) with L515 camera series.
2.1) I see in the documentation "https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet" they are all equal.
[image]
2.2) But when I use the get depth API and Point (z) it is different.
### question 
1) I would like to confirm whether the 2.2 problem was the result of the error or the wrong documentation
2) if "The same or different depth and point (z) values" then there is any mathematical formula that converts between them.
Can anybody solve the above problem for me, thank you everyone.
### lables L500
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2020-12-16:
Hi @anhTuan0712 

Is there anyway you can provide exact values that you are seeing when it comes to the depth and point (z) values being different?  Any specific example would help us be able to answer your questions.

Thanks

anhTuan0712, 2020-12-17:
> Hi @anhTuan0712
> 
> Is there anyway you can provide exact values that you are seeing when it comes to the depth and point (z) values being different? Any specific example would help us be able to answer your questions.
> 
> Thanks

Thanks for the feedback, I was wrong, the depth value was equal to the z value.

RealSenseSupport, 2020-12-21:
Hi @anhTuan0712 

One tool to look at is rs-enumerate-devices -c  (https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices)

This will provide you with the intrinsic and extrinsic of the L515 camera and provide the proper information you'll need to understand the depth with respect to other streams like RGB.

But ultimately the datasheet for L515 is correct.  The depth starting point for the L515 from an extrinsic standpoint is not on the outside of the camera but is in the inside of the camera approx. -4.5mm from the outside edge of the front cover glass.  The extrinsic if you need to take into consideration is given in this form of -4.5mm from the front cover glass.

Hope this helps.

closed 2021-01-19 by RealSenseSupport