realsenseai/librealsense #7916: Trouble building Android Example under Android Studio 4.1.1

tedlarson, 2020-12-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i                                                     | 
| Firmware Version                   |  Latest                                                     | 
| Operating System & Version |  Android 11                                            | 
| Kernel Version (Linux Only)    |  5.4                                                         | 
| Platform                                 |  ARM                                                      |
| SDK Version                            |  { 2.0 }                                                    | 
| Language                               |  {Java }                                                   | 
| Segment			        |  {Smartphone }                                      | 

### Issue Description
The Android example is built under a v3.1.1 version of Android Studio - You can't tell what the project setup options should be, and it throws errors during build about missing references.

MartyG-RealSense, 2020-12-04:
Hi @tedlarson   As a starting point in investigating your case, could you set the target SDK to '27' if you have not done so already, please?  This should help to eliminate the possibility that the problem is due to using an Android 11 device.

MartyG-RealSense, 2020-12-10:
Hi @tedlarson  Do you require further assistance with this case, please?   Thanks!

tedlarson, 2020-12-13:
This is the example I am trying to work from and I am still having issues with target SDK being 27.
https://dev.intelrealsense.com/docs/build-an-android-application-for-intel-realsense-sdk

I need to go back and follow the steps as documented here, and try to translate them to Android Studio 4.1.1.  I think the issues I am running into are more related to Android Studio setup, than SDK rev.

MartyG-RealSense, 2020-12-14:
Please do update us after your Android Studio tests.   Good luck!

MartyG-RealSense, 2020-12-21:
Hi @tedlarson   Do you have an update that you can provide, please?   Thanks!

closed 2021-01-03 by MartyG-RealSense