realsenseai/librealsense #7917: D455 can't stream in python, it shows RuntimeError: No device connected

Dontla, 2020-12-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D455 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |2.40.0.2482                        | 
| Language                            |  python                    | 
| Segment			|  {Robot }                          | 

### Issue Description
D455 can't stream in python, it shows RuntimeError: No device connected

[image]

My python code：

# -*- coding: utf-8 -*-
"""
@File    : test_D455.py
@Time    : 2020/12/4 11:13
@Author  : Dontla
@Email   : sxana@qq.com
@Software: PyCharm
"""
import pyrealsense2 as rs
import numpy as np
import cv2 as cv

pipeline = rs.pipeline()
pipeline.start()

while True:
    frames = pipeline.wait_for_frames()
    color_frame = frames.get_color_frame()
    color_image = np.asanyarray(color_frame.get_data())
    cv.imshow('win', color_image)
    cv.waitKey(0)

error：
D:\20191031_tensorflow_yolov3\python\python.exe C:/Users/Administrator/Desktop/test_D455/test_D455.py
Traceback (most recent call last):
  File "C:/Users/Administrator/Desktop/test_D455/test_D455.py", line 14, in <module>
    pipeline.start()
RuntimeError: No device connected

Process finished with exit code 1

MartyG-RealSense, 2020-12-04:
Hi @Dontla   This particular error is usually associated with how the Python wrapper was built and not a problem in the scripting.  

Are you installing the RealSense SDK using the Windows installer file from the **Releases** page and then installing the wrapper from source code with the Windows instructions on the Python wrapper page please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows

Dontla, 2020-12-04:
> Hi @Dontla This particular error is usually associated with how the Python wrapper was built and not a problem in the scripting.
>
> Are you installing the RealSense SDK using the Windows installer [...]

But I can run with D435, but not with D455 @MartyG-RealSense

MartyG-RealSense, 2020-12-04:
On a previous case of this error  on Windows where the camera worked with D435i but not with D455, building the Python wrapper with the **pip install pyrealsense2** method instead of from source code corrected the problem. 

https://github.com/IntelRealSense/librealsense/issues/7283#issuecomment-690855252

closed 2020-12-11 by Dontla