realsenseai/librealsense #7920: Question: Labview support for D455?

johnras, 2020-12-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |  NA | 
| Platform                                 | PC  |
| SDK Version                            |  2.33  | 
| Language                            |  labview  | 
| Segment			|  AR/others | 

Labview is unable to detect/open the D455, my assumption is that it is because the wrapper's api hasn't been updated to the latest version yet.  Is there as estimation of when the wrapper will be updated to support the D455?

MartyG-RealSense, 2020-12-04:
Hi @johnras   The details of the **Hello World: Color + Depth** example on the LabVIEW wrapper page reference the D455.  

[image]

I therefore would recommend replacing the **realsense2.dll** that was included in the LabVIEW package for the same file from the current version of the SDK.  You can either compile this replacement file, as indicated by the wrapper instructions, or you can obtain it from the SDK folder that is created when the full Windows SDK is set up with the installer file from the **Assets** file list on the SDK **Releases** page.

[image]

Once the SDK is installed, the pre-built **realsense2.dll** file can be found in the SDK folder's **bin** directories (32-bit in the x64 sub-directory, and 64-bit in the x64 sub-directory).

[image]

An easy way to find the SDK folder is to right-click on the **RealSense Viewer** launch icon on the Windows desktop and select **Open file location** from the menu.

The likelihood is that you will want the 64-bit version of the DLL in bin/x64.  Delete the original realsense2.dll from the LabVIEW folder and copy the new one into the folder in its place.

agrunnet, 2020-12-04:
Yes exactly

johnras, 2020-12-08:
I updated to SDK v2.40.0, copied the new realsense2.dll over to the LabVIEW folder and updated the firmware on the device to 0.5.12.09.00 and it is working!  Thank you.

closed 2020-12-08 by johnras

MartyG-RealSense, 2020-12-08:
That's great news @johnras - thanks for the update   :)

woodal, 2021-12-15:
Hellow there!
[image]

woodal, 2021-12-15:
Hello there!

[image]

I
Opening the LabVIEW Example was successful. 
Opening LabVIEW Example by changing the dll file was successful.
However, the LabVIEW program runs but no screen appears. I have both LabVIEW 64bit and 32bit. Even if I try to run both by changing the dll file, the screen does not appear.
[image]

RealSense_HelloWorld_Left_Right_IR.vi is excutable.
[image]

Any help on how to solve this problem would be appreciated.
Thanks!!

MartyG-RealSense, 2021-12-15:
Hi @woodal  Although the **Monochrome + Depth** LabVIEW example is included in the LabVIEW package, it is not described in the wrapper documentation.  From the descriptive text on the window of the image of the example that you kindly provided though, it looks as though it is streaming monochrome color from the left infrared sensor, probably with the monochrome UYVY format.

[image]

D455 supports streaming color from the left infrared imager if firmware version **5.12.8.200** or newer is installed, so your 5.13.0.50 firmware should allow this.  I do not know if this specific LabVIEW example has been tested with D455 though, as it would have more likely been designed for the older D415 model (which also supports color from the left infrared sensor - the D435 / D435i models do not).

Does the **RealSense Hello World: Color + Depth** example function correctly with your D455?  That example also streams color from the left infrared sensor.