realsenseai/librealsense #7921: D455 External Sync Signal Source Experience

jimmek, 2020-12-04:
Camera D455
FW 5.12.9.0
Windows 10
PC
SDK 2.40.0
Viewer

### Question Description
I am wanting to use an external sync signal, that is, a pulse train as described in the Intel RealSense documentation. 

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

In the documentation they used the Arduino Micro as the source of the signal. Has anyone had experience with this setup? Did it perform satisfactorily? Does anyone already have the code to run the Arduino Micro for this application or serve as a starting point? Any input as to what i may need to focus on any pitfalls to avoid?

Thank you.

RealSenseCustomerSupport, 2020-12-10:
----------------------------------------------

Hello,

The pulse/signal generator setup mentioned in the WP was used for internal demo and testing purposes and is not a supported product or device. We recommend, if possible, using a commercially available instrument which can be obtained for ~$100 or less.

We may be able to share the Arduino code, but it would be “as-is” with little or no support. Also, note that ReaslSense camera EXT Sync input needs to be 1.8V logic level. Therefore, Arduino set-up requires an additional logic level converter IC.

RS Tech Support

RealSenseCustomerSupport, 2020-12-29:
----------------------------------------------

Please let us know whether the suggestions provided are sufficient. We will close the ticket within 3 days unless we hear back.

Thanks.
RS Support

closed 2021-01-18 by RealSenseCustomerSupport