realsenseai/librealsense #7922: Hardware timestamps wrapping around can cause global timestamps to go backwards

thomas-slamcore, 2020-12-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.3.0 | 
| Operating System & Version |   Ubuntu18  | 
| Kernel Version (Linux Only)    |  5.4                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.39.0                        | 
| Language                            | C++                         | 
| Segment			|  Robot                      | 

### Issue Description

It seems that the hardware timestamps wrap around periodically, every hour or so. I've seen a commit which is meant to handle this gracefully and make sure that "global timestamps" work as expected. However I believe that the implementation is incorrect which can cause the global timestamps to go backwards.

This is the commit which attempts to address the issue - [d29667f](https://github.com/IntelRealSense/librealsense/commit/d29667fe508d67bac5bbb784c989909c6c3cbff6) from PR [#6150](https://github.com/IntelRealSense/librealsense/pull/6150).

### Unit test

Here is a unit test that demonstrates the underlying issue.

```C++
#include "catch.h"
#include <global_timestamp_reader.h>
#include <vector>

#include "catch.h"
#include <global_timestamp_reader.h>
#include <vector>

#define PERIOD_USEC 20000 // 50 Hz
#define TIMESTAMP_USEC_TO_MSEC 0.001

using namespace librealsense;

// When feeding inputs with zero latency to the linear regression
// we don't expect the linear regression to have any effect
static void testLinearCoefficients(const uint64_t ts_offset)
{
    CLinearCoefficients coeffs = CLinearCoefficients(15);

    // Create a small set of timestamps at a regular interval
    std::vector<uint64_t> timestamps;
    for (size_t i = 0; i < 10; ++i)
    {
        timestamps.push_back(ts_offset + i * PERIOD_USEC);
    }

    // Feed zero-latency "ping" samples
    for (const auto ts : timestamps)
    {
        // Create matching hardware and system timestamps
        const double hardware_ts = (ts % UINT32_MAX) * TIMESTAMP_USEC_TO_MSEC;
        const double system_ts = ts * TIMESTAMP_USEC_TO_MSEC;

        // Feed sample for linear regression
        coeffs.update_samples_base(hardware_ts);

        CSample sample(hardware_ts, system_ts);
        coeffs.add_value(sample);
    }

    // It should be possible to query timestamps in the past
    // This is to account for a delay before a frame or packet is received
    for (const auto ts : timestamps)
    {
        // Create matching hardware and system timestamps
        const double hardware_ts = (ts % UINT32_MAX) * TIMESTAMP_USEC_TO_MSEC;
        const double system_ts = ts * TIMESTAMP_USEC_TO_MSEC;

        // Query system time at the point of the sample
        coeffs.update_samples_base(hardware_ts);

        coeffs.update_last_sample_time(hardware_ts);
        const double queried_ts = coeffs.calc_value(hardware_ts);

        // We fed matching timestamps so we expect the output to be equal
        REQUIRE(queried_ts == system_ts);
    }
}

TEST_CASE("linear_coefficients_simple", "")
{
    testLinearCoefficients(0);
}

TEST_CASE("linear_coefficients_timewrap", "")
{
    // Start a little bit before the hardware wrap-around time
    const uint64_t ts_offset =
      ((UINT32_MAX - 5 * PERIOD_USEC) / PERIOD_USEC) * PERIOD_USEC;

    testLinearCoefficients(ts_offset);
}
```

RealSenseSupport, 2021-04-06:
@thomas-slamcore I saw you've submitted PR https://github.com/IntelRealSense/librealsense/pull/7923 and waiting to be merged. Can we close this issue and track PR accordingly? Looking forward to your update. Thanks!

thomas-slamcore, 2021-04-06:
If that´s what you would normally do, then go ahead. I´m mainly concerned that there doesn´t seem to be a review forthcoming.

RealSenseSupport, 2021-04-09:
@thomas-slamcore Thanks for your feedback and contribution! Close this accordingly.

closed 2021-04-09 by RealSenseSupport