realsenseai/librealsense #7924: No RealSense devices were found for L515 on RealSense ROS

eltonhcchan, 2020-12-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515 } | 
| Firmware Version                   | 01.05.02.00| 
| Operating System & Version |   {Win 10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.40.0.2482}                          | 
| Language                            |  {C/pcl}                          | 
| Segment			|  {others }                          | 

### Issue Description
I've installed ROS on Windows (Noetic), RealSense ROS v2.2.20 as well as librealsense. However, when tried to set a node for the camera, it cannot locate the test_nodelets.xml file and subsequently, cannot find the camera. Although I installed the latest RS SDK that comes with librealsense v2.40.0, Realsense ROS was still built with librealsense v2.34.0. (See the log below)

L515 is working properly with RealSense Viewer and can be identified using rs-enumerate-devices, and thus there is no connection problem.

```
c:\catkin_ws>roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5748]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to 2602a8d1-370f-11eb-93c9-af3e09eb46ed
process[rosout-1]: started with pid [7744]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [4928]
process[camera/realsense2_camera-3]: started with pid [11448]
[ERROR] [1607182380.594243600]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1607182380.594609400]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet/test_nodelet.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1607182380.611082500]: Initializing nodelet with 8 worker threads.
[ INFO] [1607182380.945155700]: RealSense ROS v2.2.20
[ INFO] [1607182380.945473900]: Built with LibRealSense v2.34.0
[ INFO] [1607182380.946694600]: Running with LibRealSense v2.34.0
[ WARN] [1607182381.040177000]: No RealSense devices were found!
[ WARN] [1607182387.079765900]: No RealSense devices were found!

```

RealSenseSupport, 2021-03-24:
Hi,

Will you be needing further help with this?

Thanks

closed 2021-04-02 by RealSenseSupport

zsm666, 2022-04-07:
how to solve this problem?my librealsense is V2.5,ROS is melodic.QAQ