realsenseai/librealsense #7926: Create a sensor from Software_device who can not call Run and Open function

DragonDeux, 2020-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

DragonDeux, 2020-12-06:
Dear Realsense engineers:

I'm trying to sync non-realsense RGB device with Realsense Lidar L515 device by following code:

```
sync = rs.syncer
colorSensor = rs.software_device().add_sensor("Color")
balserColorSensor.open(color_stream)
balserColorSensor.start(sync)
```
error: 
**AttributeError: 'pyrealsense2.pyrealsense2.software_sensor' object has no attribute 'open'**

Btw, i have inspired this kind of code with :https://github.com/IntelRealSense/librealsense/blob/master/examples/software-device/rs-software-device.cpp which is coded in C++, i thought it would work bec it is Realsense official sample.

Can anyone please help to resolve this issue? 

Thanks in advance

RealSenseSupport, 2021-03-24:
Hi,

Will you be needing further help with this? While we cannot assist with 3rd party sensors, we tested the latest rs-software-device.cpp example and it appears to be working as expected. As an aside I see you are getting a python error. The python wrapper may not be implementing the exact interface available in the CPP SDK so we'd recommend using the API directly to avoid such potential issues.

Thanks

*FYI if we don’t hear from you in 7 days, this issue will be closed.*

closed 2021-04-02 by RealSenseSupport