realsenseai/librealsense #7927: The Librealsense installation instructions for Raspbian are extremely dated

TCIII, 2020-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R435i 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Raspbian Buster 
| Kernel Version (Linux Only)   |  4.19                                         | 
| Platform                                |Rpi 3B+                                   |                                                                                                                   
|SDK Version                           |  2.40                          | 
| Language                             | Python 3.6                         | 
| Segment			       | Robot                          | 

### Issue Description
The Title says it all. Example: Protobuf is now at version 3.14.

MartyG-RealSense, 2020-12-07:
Hi @TCIII  You have a valid point.  More recently updated Pi guides such as the Acrobotic one for Pi 4 (which is structured very similarly to Intel's official Raspbian guide) install Protobuf 3.10.0.  

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

My understanding though is that so long as the Protobuf is a Version 3 number then it should work.

TCIII, 2020-12-07:
Protobuf v3.14 no longer contains "autogen.sh" which causes protobuf installation issues. Deprecated versions of protobuf are in a different directory,

MartyG-RealSense, 2020-12-07:
The Raspbian instructions are likely to remain in their current form for the foreseeable future.

closed 2020-12-07 by TCIII