realsenseai/librealsense #7928: Why D415 camera is mounted on the USB 2.0 driver instead of USB 3.0 of the ARM platform?

hywang-66, 2020-12-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Linux  | 
| Kernel Version (Linux Only)    |  4.9   | 
| Platform                                 | ARM |
| SDK Version                            |  2.35.0  | 
| Language                              | C++ | 

### Issue Description 

When camera is plugged into the ARM platform, why does it show it is mounted under the USB 2.0 instead of USB 3.0? And how to mount the camera under USB 3.0? Thank you!  

The current USB port of ARM platform has two drivers, USB 3.0 and USB 2.0 respectively. When using the original USB Type-C cable to connect the camera to the ARM platform, the camera is mounted under the USB 2.0 driver instead of USB 3.0. When connect to the USB 3.0 of PC, it will be mounted under the USB 3.0. No change after re-plugging several times. The following figures show the situation on the PC and ARM platforms respectively.  

[image]
<p align="center">connect the camera to ARM</p>  

[image]
<p align="center">connect the camera to PC</p>

MartyG-RealSense, 2020-12-07:
Hi @hywang-66   It looks as though you are using Arm for Ubuntu (the version of Ubuntu for Arm).

Could you tell me please the name of the Arm device that you are attaching the camera to?  This is important, as Arm Platform can literally mean a device that has an Arm processor.  Some Arm devices, such as Raspberry Pi boards before Pi 4, only have a USB 2 port and so would not be able to provide USB 3.

hywang-66, 2020-12-07:
@MartyG-RealSense 
The arm platform is Zynq Ultrascale+ which has USB 3.0 port.

MartyG-RealSense, 2020-12-07:
What method are you using to build librealsense please - from DKMS packages or by building from source code?

If it is from packages then that could explain why it works on your PC but not Arm.  This is because the DKMS packages are designed for computers with x86 or x64 architecture processors, whilst your Zynq device has Arm architecture.  The DKMS packages would not work with Arm.

Also, librealsense does not officially support Kernel 4.9.  It goes from 4.8 to 4.10.

[image]

Instead, I would recommend building librealsense from source code using the **RSUSB backend** installation.  This method requires an internet connection for the installation process but it is not dependent on Linux versions or kernel versions and does not require patching.  Instructions for this process, which is enabled by adding the customization flag **-DFORCE_RSUSB_BACKEND=true** to a CMake build instruction, can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

hywang-66, 2020-12-08:
@MartyG-RealSense 
I try you advice, then re-compile the library again. But it is still mounted on the USB 2.0 driver.

MartyG-RealSense, 2020-12-08:
Thank you very much.  Could you run **rs-enumerate-devices** and post the output here please?

Also, if you have access to the **RealSense Viewer** program, does it say '2.1' or '3.2' beside the camera name at the top of the options side-panel?  2.1 indicates USB 2 and 3.2 indicates USB 3.  Whilst in the Viewer, you could go to the **More** option at the top of the side-panel and select **Hardware Reset** from its drop-down menu to reset the camera without unplugging and replugging the camera.  If you can do so, does the USB status change?

[image]

hywang-66, 2020-12-08:
@MartyG-RealSense 
Sorry, my Linux system currently only has a terminal and no UI. So I can not run RealSense-Viewer. Whether this issue is related to USB3.0 compatibility？

MartyG-RealSense, 2020-12-08:
If your Linux system is running in a "headless" setup without a display, is it also installed on a Virtual Machine (VM)?  This can cause problems with USB, as described in the opening section of the installation instructions for building librealsense from source code.

"Due to the USB 3.0 translation layer between native hardware and virtual machine, the librealsense team does not support installation in a VM.  If you do choose to try it, we recommend using VMware Workstation Player, and not Oracle VirtualBox for proper emulation of the USB3 controller".

hywang-66, 2020-12-09:
@MartyG-RealSense 
I put the Linux image file in the memory. When the system is powered on, ARM will load the Linux image file in the memory. Mainly use the serial port on ARM to interact with Linux system. Do you mean this image file is the same as the Virtual Machine？

MartyG-RealSense, 2020-12-09:
Your project's setup does sound as though it has characteristics of a Virtual Machine (emulating a computer system, a Linux system in this case).  

https://en.wikipedia.org/wiki/Virtual_machine

hywang-66, 2020-12-10:
@MartyG-RealSense 
Do you have any good suggestions on which board can be used to make the Realsense camera work under USB3.0, and can output 640*480p60 frames of data stably. The board is required to be small in size and relatively inexpensive, and it can be displayed using interfaces such as VGA or HDMI port. Thank you!

MartyG-RealSense, 2020-12-10:
**Raspberry Pi 4 Model B** should meet all of those requirements.  The Pi 4, which is an Arm board, has been widely used with RealSense 400 Series cameras.

https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

hywang-66, 2020-12-10:
@MartyG-RealSense Thank you!!!

MartyG-RealSense, 2020-12-17:
Hi @hywang-66  Do you require further assistance with this case, please?  Thanks!

hywang-66, 2020-12-21:
@MartyG-RealSense It's enough. Thanks!

closed 2020-12-21 by MartyG-RealSense