realsenseai/librealsense #7930: How to get Depth_z16 from Color image

mookor, 2020-12-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Linux Ubuntu 16  |                        | 
| Language                            |  python                        | 
| Segment			|  others                           | 

### Issue Description

I am trying to reconstruct a depth image from a color image.
I did everything according to this guide https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras
But the result is a GrayScale image, but how to make z16 out of it, so that you can use methods such as, for example, get_distance

MartyG-RealSense, 2020-12-07:
Hi @mookor   Section 3.2 of the **depth image compression by colorization** white paper states that recovery of a **colorized** depth image is not currently available in the RealSense SDK, and so it has to be accomplished with an ordinary C++ script that is independent of the SDK.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-3-2-depth-image-recovery-from-colorized-depth-images-in-c

The script provided in the paper is for C++ though.  The paper does not provide a Python equivalent but explains that a version of the script for other langiages can be created if the same logic that is in the C++ script is applied.   My research was not able to find a pre-made Python script at this time however, so it is a conversion that you will have to undertake yourself.

mookor, 2020-12-07:
@MartyG-RealSense  I wrote my script in python that does the translation of their color image into depth, but I get a (640x480x1) grayscale image, to which it is impossible to apply methods such as, for example, get _distance, how can this be done?

MartyG-RealSense, 2020-12-07:
Apologies for the delay in responding further.  As recommended in the depth colorization paper beneath the C++ script, have you ensured that the values of min_depth and max_depth are the same as those that were used for colorization?

Also, would it be possible to post the code of your Python version of the recovery script in this discussion please?  Thanks for your patience.

MartyG-RealSense, 2020-12-14:
Hi @mookor   Do you require further assistance with this case, please?  Thanks!

mookor, 2020-12-15:
hello, sorry for the long answer!
I am able to accurately restore the greyscale image, but I do not know how to get the distance from there, I can not find the formula for this function "get_distance"

below I have provided the code how I was recovering the image
```

@numba.njit
def RGBtoD(r, g, b):
    if r + g + b < 255:
        return 0
    if r >= g and r >= b:
        if g >= b:
            return g - b
        else:
            return (g - b) + 1529

    elif g >= r and g >= b:
        return b - r + 510
    elif b >= g and b >= r:
        return r - g + 1020

@numba.njit
def ColorToD(frame):
    min_depth = 0
    max_depth = 10
    depth_frame = np.zeros((frame.shape[0], frame.shape[1]))
    div = max_depth - min_depth
    for i in range(frame.shape[0]):
        for j in range(frame.shape[1]):
            r, g, b = frame[i][j][0], frame[i][j][1], frame[i][j][2]
            hue_value = RGBtoD(r, g, b)
         
            z_value = (min_depth + ((div) * hue_value) / 1529)
            depth_frame[i][j] = z_value / 255
            
    return depth_frame`
```

MartyG-RealSense, 2020-12-15:
There is a Python script that uses get_distance in the link below.   I hope that you will find some useful insights in it to apply to your own script.

https://github.com/IntelRealSense/librealsense/issues/5878

mookor, 2020-12-16:
@MartyG-RealSense  You do not understand me(
The fact is that I get a NumpyArray, and the get_distance () method applies only to objects of the z16 format, so I'm looking for information on how to find the distance from the NumpyArray

MartyG-RealSense, 2020-12-16:
It sounds as though you want to convert a numpy array to librealsense depth frame.  Others have attempted this but as far as I know there is not a definitive solution for doing so at the time of writing this.   

https://support.intelrealsense.com/hc/en-us/community/posts/360047819294-Conversion-of-Frame-to-Numpy-Array-and-Vice-Versa

mookor, 2020-12-18:
not really, I just want to know how, for example, from such an image (numpy array) to get the distance for each point of the image
[https://habrastorage.org/files/b42/18b/f3d/b4218bf3d5794c2d9ee13797920862c6.png](url)

mookor, 2020-12-18:
Not certainly in that way.
I just need to find out how I can get the distance values ​​for each point in the image from a black and gray image.
Sample image
https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQ-3FX91jXkKKn-07DS7JWRoWkxAraFD2LVyA&usqp=CAU

MartyG-RealSense, 2020-12-18:
I carefully considered your needs.   I wonder if the discussion below regarding getting depth from 1-channel CV_32FC1 would meet your requirements.

https://github.com/IntelRealSense/librealsense/issues/6101

MartyG-RealSense, 2021-01-01:
Case closed due to no further comments received.

closed 2021-01-01 by MartyG-RealSense

gywhitel, 2022-12-14:
> Hi @mookor Section 3.2 of the **depth image compression by colorization** white paper states that recovery of a **colorized** depth image is not currently available in the RealSense SDK, and so it has to be accomplished with an ordinary script that is independent of the SDK.
> 
> https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-3-2-depth-image-recovery-from-colorized-depth-images-in-c
> 
> The script provided in the paper is for C++ though. The paper does not provide a Python equivalent but explains that a version of the script for other languages can be created if the same logic that is in the C++ script is applied. My research was not able to find a pre-made Python script at this time however, so it is a conversion that you will have to undertake yourself.

Hi, I think there should a 1529 in the Equation 2. for the inverse colorization method.