realsenseai/librealsense #7933: Can not align frameset after post-processing of frameset

DragonDeux, 2020-12-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

DragonDeux, 2020-12-07:
```
........<configuration>
align_to = rs.stream.color
align = rs.align(align_to)        
frameset = pipeline.wait_for_frames()
newFrameset = temp_filter.process(frameset)
aligned_frames = align.process(newFrameset)
```
Error: Invoked with: <pyrealsense2.pyrealsense2.align object at 0x00000279FF46DFB0>, <pyrealsense2.frameset Z16 Y8 BGR8 #1>

Dear Realsense engineers,

i have encounter a issue about using post-processing filter of depth map before aligning depth-RGB frame, unfortunately, it reports a error above mentioned, it is recommended to proceed post-filter processing before doing alignment isn't it?  many appreciate 

regards

MartyG-RealSense, 2020-12-07:
Hi @DragonDeux   The link below has a simple Python example of using align with post-processing:

https://github.com/IntelRealSense/librealsense/issues/2356#issuecomment-465766323

Yes, it is recommended to perform align **after** post-processing in order to help to avoid a distortion effect called **aliasing** that can cause jagged lines.

If you would like to further explore implementing post-processing filters in Python, Intel have published a Python tutorial about the subject.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

MartyG-RealSense, 2020-12-21:
Case closed due to no further comments received.

closed 2020-12-21 by MartyG-RealSense

dohesaat, 2023-01-12:
I have a related problem / question. Not sure whether i would open a new issue?
My problem is that I get a misalignment between the color and depth information of a frameset when I call
`frameset = depth_to_disparity.process(frameset) # depth_to_disparity = rs.disparity_transform(True)
frameset = disparity_to_depth.process(frameset).as_frameset() # disparity_to_depth = rs.disparity_transform(False)
`
before alignment. More specifically, I get a shift of a few cm between the (not) aligned depth and color frame.
another issue is that i need to init the aligner
`aligner = rs.align(rs.stream.color)`
within the while loop when the aligner.process call occurs after the post processing (in this case the disparity transform).
When I first allign and then do some post processing, initializing the aligner once (outside the while loop for streaming frames) is enoguh as expected.

thanks for you help, dom

MartyG-RealSense, 2023-01-12:
Hi @dohesaat  Intel recommend placing the align process **after** post-processing filters if you are not doing so already.  This helps to avoid distortion effects such as aliasing (jagged lines).

`aligner = rs.align(rs.stream.color)` should be outside of the while loop (just above where the while loop begins), whilst the `align.process` instruction should be inside the while loop.

This principle is demonstrated by the SDK's Python example program 'align_depth2color.py'.

**Outside while loop**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L60-L61

**Inside while loop**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L65-L71

The list of post-processing filters should be placed after the pipe start instruction but does not have to be inside of a while loop, as demonstrated by the Python post-processing code snippet at https://github.com/IntelRealSense/librealsense/issues/1672#issuecomment-387438447