realsenseai/librealsense #7934: How to get Left and Right image pairs from RealSense D455

changchenliang, 2020-12-07:
Hi. I am using the D455 camera on my Arm Platform (RK3399 + Linux + Python 3.7).  
In my work I need to get Left and Right image pairs in python from the D455 camera. However, when connected to the ARM platform, I can only capture one RGB image and one IR image (with dots pattern on it). So my question is how to get Left and Right image pairs (RGB image pairs or clean IR image pairs without laser dots are both OK for me) ?
Actually the clean left and right raw IR image pairs can be shown in RealSense Viewer app by changing settings in "Stereo depth" item, but when I close RealSense Viewer it can not remember this settings, is there any way to permanently save/write this settings to the camera in order for me to capture two IR images in python using opencv? ..    
Thank you for your time to my questions

MartyG-RealSense, 2020-12-07:
Hi @changchenliang   To access both the left and right IR streams with scripting, you should give each stream its own index number in order to differentiate between the two.  The discussion in the link below has an example Python script for doing so.

https://github.com/IntelRealSense/librealsense/issues/3878#issuecomment-569550710

If you find that you need to control the on-off state of the IR emitter with Python, the resources in this link should be helpful:

https://github.com/IntelRealSense/librealsense/issues/7919#issuecomment-738857679

If you need to make use of OpenCV, the discussion linked to below is for C++ but contains cv code for the IR frames that could be converted to your Python project.

https://github.com/IntelRealSense/librealsense/issues/1480

changchenliang, 2020-12-08:
Thank you for the reply. I come across with another question: when I try to build python wrapper, I have an error and can not finish. The details are below:
my steps are: 
1. first build librealsense from source (success)
2. then separately build python wrapper. 
however, when I come to the step of: make -j4 (or make -j8), there is an error shown below: 
C++: fatal error
 
[image]

Seems it happens when building realsense-viewer. So how can I solve this problem in order to build python wrapper? 

Thanks!

changchenliang, 2020-12-08:
> Hi @changchenliang To access both the left and right IR streams with scripting, you should give each stream its own index number in order to differentiate between the two. The discussion in the link below has an example Python script for doing so.
> 
> [#3878 (comment)](https://github.com/IntelRealSense/librealsense/issues/3878#issuecomment-569550710)
> 
> If you find that you need to control the on-off state of the IR emitter with Python, the resources in this link should be helpful:
> 
> [#7919 (comment)](https://github.com/IntelRealSense/librealsense/issues/7919#issuecomment-738857679)
> 
> If you need to make use of OpenCV, the discussion linked to below is for C++ but contains cv code for the IR frames that could be converted to your Python project.
> 
> #1480

Thanks to your reply. However, when I try to use import pyrealsense2 as rs, I have a problem:

when I run the code and it comes to pipeline.start(config), there is an error "Runtime error: no device connected", I am using D455 on a 3399pro ARM platform. 
How to solve this problem?

MartyG-RealSense, 2020-12-08:
May I ask which method you used to build the Python wrapper please?  (pip install, or building the wrapper from source code or building librealsense and the wrapper from source code at the same time)

On the last occasion that there was a Python issue on an RK3399 system, Intel advised building the wrapper from source code.

https://github.com/IntelRealSense/librealsense/issues/7424#issuecomment-705945552

changchenliang, 2020-12-09:
> May I ask which method you used to build the Python wrapper please? (pip install, or building the wrapper from source code or building librealsense and the wrapper from source code at the same time)
> 
> On the most recent occasion that there was a Python issue on an RK3399 system, Intel advised building the wrapper from source code.
> 
> [#7424 (comment)](https://github.com/IntelRealSense/librealsense/issues/7424#issuecomment-705945552)

Thank you for your help. Actually it's solved now. I build the librealsense and wrapper from the source code at the same time. I think the problem is because I did not do some of the pre-requisites on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
After I did the pre-requisites, it works without errors! 

Now I just come to the last step, and find another problem. When I use python script to capture the left IR image under color "RGB8" format, the image looks strange. The white scene appears very pink and not correct, as shown below: 
 
[image]

The right image is captured from the RGB camera which is normal, but in the left image captured from the left IR camera under "rgb8" format, the white lamp looks very pink and is not correct. (I find that when somewhere is very bright, it will look very "pink" in the image...) 

So how to solve this and set parameters in python script to correct the IR camera stream in color? 

Thanks.

changchenliang, 2020-12-09:
Thank you for your help. Actually it's solved now. I build the librealsense and wrapper from the source code at the same time. I think the problem is because I did not do some of the pre-requisites on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
After I did the pre-requisites, it works without errors! 

Now I just come to the last step, and find another problem. When I use python script to capture the left IR image under color "RGB8" format, the image looks strange. The white scene appears very pink and not correct, as shown below: 
[image]

The right image is captured from the RGB camera which is normal, but in the left image captured from the left IR camera under "rgb8" format, the white lamp looks very pink and is not correct. (I find that when somewhere is very bright, it will look very "pink" in the image...) 

So how to solve this and set parameters in python script to correct the IR camera stream in color? 

Thanks.

changchenliang, 2020-12-09:
> May I ask which method you used to build the Python wrapper please? (pip install, or building the wrapper from source code or building librealsense and the wrapper from source code at the same [...]

Thank you for your help. Actually it's solved now. I build the librealsense and wrapper from the source code at the same time. I think the problem is because I did not do some of the pre-requisites on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
After I did the pre-requisites, it works without errors! 

Now I just come to the last step, and find another problem. When I use python script to capture the left IR image under color "RGB8" format, the image looks strange. The white scene appears very pink and not correct, as shown below: 

The right image is captured from the RGB camera which is normal, but in the left image captured from the left IR camera under "rgb8" format, the white lamp looks very pink and is not correct. (I find that when somewhere is very bright, it will look very "pink" in the image...) 

So how to solve this and set parameters in python script to correct the IR camera stream in color? 

Thanks.
[image]

MartyG-RealSense, 2020-12-09:
The pink-purple hue is expected for an RGB8 image from the left IR sensor.  The image below compares RGB from the RGB sensor (upper) to RGB8 from the Left IR sensor (lower) on the D455.

[image]

The D415 model, which also supports RGB8 from the left IR sensor, renders color smilarly.

[image]

If you need left and right IR images though, Y8 monochrome would likely be the best option.  This is because the D455 and D415 can only output color such as RGB8 from the left IR sensor.  Otherwise, if you need matching IR images then use Y8 IR for both left and right and use RGB for color.

changchenliang, 2020-12-09:
Got it. So I will just use the left and right IR sensor in y8 mono type. Thank you for your time to help me. 
My case can be closed.

MartyG-RealSense, 2020-12-09:
Thanks very much @changchenliang

closed 2020-12-09 by MartyG-RealSense