realsenseai/librealsense #7935: The imu data after calibration(D435i)

Phz-1995, 2020-12-07:
Hello,

I'm planning to get the imu data (D435i) from realsense-ros. I have done the imu calibration using the official script rs-imu-calibration.py and save the result to device. My question is: Is the data I got from realsense-ros calibrated? As in #7103 mentioned, there is a Motion correction button in realsense viewer, which can correct the IMU data. I have no idea if this correction is based on the calibration result and this correction is only limited in realsense viewer and not included in realsense-ros. In that case, the data I got from realsense-ros is still the original one and the calibration is meaningless.

Thanks.

MartyG-RealSense, 2020-12-07:
Hi @Phz-1995  Performing an IMU calibration and writing it to the camera is a very useful action.  The IMU component used in the RealSense range does not have an internal calibrator, and so if it is uncalibrated then its acceleration (accel) may not settle at zero even when the camera is stationary.  

Once a calibration has been written to the camera then motion correction can be accessed.  Also, Y-accel should be much closer to the ideal target of **9.80** (9.81 represents the Earth's gravity).  

There is not a direct counterpart for IMU motion correction in the RealSense ROS wrapper, however.  Intel's guide to SLAM in ROS with the D435i is an excellent reference for learning about using the IMU in Realense ROS.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

Phz-1995, 2020-12-07:
> Hi @Phz-1995 Performing an IMU calibration and writing it to the camera is a very useful action. The IMU component used in the RealSense range does not have an internal calibrator, and so if it is uncalibrated then its acceleration (accel) may not settle at zero even when the camera is stationary.
> 
> Once a calibration has been written to the camera then motion correction can be accessed. Also, Y-accel should be much closer to the ideal target of **9.80** (9.81 represents the Earth's gravity).
> 
> There is not a direct counterpart for IMU motion correction in the RealSense ROS wrapper, however. Intel's guide to SLAM in ROS with the D435i is an excellent reference for learning about using the IMU with ROS.
> 
> https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
> 
> There was a recent discussion about how **RTABMAP** could be used with RealSense and ROS for correction of drift.
> 
> [IntelRealSense/realsense-ros#1529](https://github.com/IntelRealSense/realsense-ros/issues/1529)

Thanks for the quick response, I just want to confirm if the data i got from ROS wrapper will change after i have done a IMU calibration?

MartyG-RealSense, 2020-12-07:
My understanding is that ROS uses raw IMU data rather than 'fixed' data like that provided in librealsense by motion correction.  And raw IMU data can be noisy.  So whilst I would always recommend having the IMU calibrated, you may not see as much of a difference to the results in ROS as you would in librealsense applications with motion correction.

Phz-1995, 2020-12-07:
> My understanding is that ROS uses raw IMU data rather than 'fixed' data like that provided in librealsense by motion correction. And raw IMU data can be noisy. So whilst I would always recommend [...]

I‘m not sure I really understand what you said, do you mean the raw data provided by ROS is the calibrated one, although there is still some noise which can not be eliminated in ROS but can be eliminated in librealsense "motion correction" function ?

MartyG-RealSense, 2020-12-07:
The RealSense ROS wrapper operates independently from librealsense.  Settings made in librealsense programs do not affect the camera's performance in the ROS wrapper, and vice versa.   

Motion correction is done in software that supports that function, such as the RealSense Viewer or the **rs-motion** example.  Therefore in ROS, you can benefit from a good IMU calibration being written to the camera hardware by the Python IMU calibration tool.  But you will not benefit from **additional** software-based assistance from motion correction algorithms.

In short, you may experience noiser data in the RealSense ROS wrapper than when the camera is used with librealsense programs that support motion correction and customizing the sensitivity of the IMU.

Phz-1995, 2020-12-08:
> The RealSense ROS wrapper operates independently from librealsense. Settings made in librealsense programs do not affect the camera's performance in the ROS wrapper, and vice versa.
> 
> Motion correction is done in software that supports that function, such as the RealSense Viewer or the rs-motion example. Therefore in ROS, you can benefit from a good IMU calibration being written to the camera **hardware** by the Python IMU calibration tool. But you will not benefit from additional **software** assistance from motion correction algorithms.
> 
> In short, you may experience noiser IMU data in the RealSense ROS wrapper than when the camera is used with librealsense programs that support motion correction and customizing the sensitivity of the IMU.

That's very clear, thank you!

closed 2020-12-08 by Phz-1995