realsenseai/librealsense #7936: Realsense L515 Hardware Sync issue (Illegal HW state)

aakshfs, 2020-12-07:
|  Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Ubuntu 18.04.5  | 
| Kernel Version (Linux Only)    |  5.4.0-53-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.39.0  | 
| Language                            |  Python/C++  | 

Hello!

I am working on a project requiring two L515 to use hardware sync. I have followed the paper posted [here](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) but unfortunately the camera doesn't turn on when receiving a high input on the HW sync pins (the camera is set to HW sync mode, but the streams just times out after 5 seconds with the error `Frame didn't arrive within 5000`).

Moreover, when trying to switch to inter cam sync mode from `realsense-viewer` while the camera is already running, I get the error: 
```
Invalid Value in rs2_set_option(options:0x7f6f7001c760, option:Inter Cam Sync Mode, value:1): hwmon command 0x19( 1 0 0 0 ) failed (response -38= Illegal HW state)
```
If I stop the stream, set the camera to HW sync mode and start it again, I get the same timeout while waiting for frames error.

What could be causing this behaviour? Is there a way to check the connection of the connector to the camera (for example, expected the resistance across pin 1 and 2)? 
Is there a way to monitor the HW sync pins (preferably from software), so I know for sure when/if the camera receives a high signal?

kkh3049, 2020-12-07:
We're working with several L515s as well, and have been debugging our HW Sync which we set up based on the page you linked but in C#.  In the last 30 minutes, we found that the L515's were enabled when our sync signal was Low, not when our sync signal was High. (Opposite to the Raspberry Pi example code they gave).  Not sure why that is, whether a wiring mixup or a mis-documentation, but you could try flipping your highs and lows, and see if that gets you what you'd expect.

aakshfs, 2020-12-07:
Unfortunately I have tried alternating the high/low signal in 500ms intervals, and also starting the camera with a high/low signal with no luck.

It seems pretty weird that the issue was the same for both cameras in your case. I haven't seen a lot of people (if any) posting about this, so if there's a mixup, I guess there are a few cameras out there with the same issue.

aakshfs, 2020-12-08:
After a bit of experimenting, I found that the connection to the HW sync pins of the camera was wrong. There should be a 2M ohm resistance across the pins, if you want to check it is connected.
Also, I also had the same issue about the cameras being on while the HW sync signal is low (opposite to the example in the paper). So maybe there was a mistake in the paper.

closed 2020-12-08 by aakshfs

kkh3049, 2020-12-08:
@aakshfs 
If you're willing, can you share your code for activating the Hardware Sync on the RealSense SDK side?  We're still struggling to get our sync working programatically?