realsenseai/librealsense #7937: rs-capture Example not working

FinnMckinn, 2020-12-07:
Hi all,

I am currently attempting to go through the examples provided at https://dev.intelrealsense.com/docs/code-samples though in 'rs-capture' the provided code does not seem to work. When run, the GLFW window opens but just stays white and unresponsive when there should be two images on it. I've stepped through the code and everything seems to be working, no bad values and no errors.

Any help would be appreciated.

MartyG-RealSense, 2020-12-07:
Hi @FinnMckinn   As a starting point in investigating your case, could you tell me please whether you are able to launch the **RealSense Viewer** program successfully please?  As the Viewer relies on GLFW, if it can launch then it would suggest that GLFW is working correctly on your computer.

FinnMckinn, 2020-12-07:
@MartyG-RealSense I am able to launch and use RealSense Viewer program. I am currently running the program on a Windows 10 machine with a RealSense 435i.

After going through the code from the website again I am now having an error thrown at me: "Rendering is currently supported for video, motion and pose frames only".

MartyG-RealSense, 2020-12-07:
Are you running rs-capture from the SDK **Tools** folder?  This is found by right-clicking on the Viewer's launch icon on the Windows desktop and selecting the **Open file location** menu option.  Then find the **rs-capture.exe** file and double-leftclick on it to launch the program.

If you are already launching rs-capture from the Tools folder, are you able to successfully launch **rs-multicam**?  Like capture, this should display a set of panels of the D435i streams.

FinnMckinn, 2020-12-07:
@MartyG-RealSense Using this method gets it to run, I didn't know the code was already in the library. Thank you for the information, using I should be able to figure out the problem.

closed 2020-12-07 by FinnMckinn

MartyG-RealSense, 2020-12-08:
I'm pleased to have been able to help.   :)