realsenseai/librealsense #7939: L515 Hardware Synchronization Clarifications

kkh3049, 2020-12-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | 1.5.2.0 | 
| Operating System & Version |   Win10 : 19041.630  | 
| Kernel Version (Linux Only)    |  N/A                                       | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  2.36.0.0 }                          | 
| Language                            |  C#/Python                          | 

### Issue Description
We've been trying to replicate the results from the whitepaper [Multi-Camera configurations with the Intel® RealSense™ LiDAR Camera L515](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) using a C# to drive 8 L515s from a PC and Python on a Raspberry Pi to drive the Hardware Sync signals, but haven't gotten the results we expect, and would like some further clarification regarding how it works.
1. We recognized that the example Python code is run only once, so we wrapped setting GPIO output signals in a loop and had the script run as a service continually as long as the Raspberry Pi is running. I'm assuming those behind the whitepaper did similarly?  If not, how did they have the Raspberry Pi signal output at an appropriate time?
2. We also opened all 8 of the L515's depth streams in the RealSense viewer with Intercam Sync Mode and found that our L515's were streaming when the sync signal was LOW not when the sync signal was HIGH, so we switched the two in our code.  Is this the intended sync signal on the L515's, where LOW is enabling streaming?  If so, it contradicts the image and associated explanation given in the whitepaper in [section 2-E](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#section-e-external-trigger): [image] And if not, then I must have misunderstood the wiring given in [section 1](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#section-1connecting-the-cameras).  Which pin (top, middle or bottom) in the following image corresponds to the sync signal input, and which to the grounding cable? [image] (We interpreted the middle as the sync cable and the bottom as the ground cable.)
3. Once we were able to see Hardware Syncing in the RealSense Viewer we attempted to utilize HW syncing programmatically in C#.  I notice that in the whitepaper ([section 3-A](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#section-a-programming-trigger-configuration-prior-to-streaming)), they set inter cam hardware sync on the first camera (through using `front()`).  Is this the recommended way to set this?  Or should we be searching for the depth camera specifically with something like this:
```C#
var m_pipe = new Pipeline(ctx);
var cfg = new Config();
cfg.EnableDevice(m_serialNumber);
cfg.EnableStream(Stream.Depth, 640, 480);
cfg.EnableStream(Stream.Color, 1280, 720, Format.Bgr8, 6);

var pipelineProfile = cfg.Resolve(m_pipe);
var sensors = pipelineProfile.Device.QuerySensors();
foreach (var sensor in sensors)
{​​​​​​​​
    if (sensor.Is(Extension.DepthSensor) || sensor.Is(Extension.L500DepthSensor))
    {​​​​​​​​
        sensor.Options[Option.InterCamSyncMode].Value = 1.0f;
        break;
    }​​​​​​​​
}​​​​​​​​
```
Further, is this the appropriate method of setting the HW Sync from C#, or is there another preferred method? This hardware sync is important for our product, and would really appreciate any guidance/clarifications about this process.

(On another note, is there a good example in ensuring we get color images that match the depth images?  We really only want a single still frame from all 8 cameras, without interference.  How does the Hardware Sync alter the synchronization between color images and depth images?)

kkh3049, 2020-12-08:
We have not been successful in activating Hardware sync programatically.  Any help would be appreciated.

jlscaife, 2020-12-14:
Hi @kkh3049 

Let me see if I can help you in your questions above:

- Initial question regarding Python, yes our initial proof of concept in the whitepaper was only centered around Python and used in the same manner you describe.  We haven't ported or used any other language or wrapper.
- Regarding the signals, I think you do have the wires mixed up.  The behavior is as we describe in the whitepaper.  We also made a couple of edits to the whitepaper to hopefully clarify which pins on the connector are pin 1, 2, 3 and which one is the sync signal and which one is ground (https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup).  Have a look there but the clarify here

[image]
The left most pin is pin 1 for sync signal (or another way to look at it, exactly in the image above, the bottom pin is pin 1 for sync signal).  Pin 2 is the middle and that is ground.  The right most pin is pin 3 (or another way to look at it, exactly in the image above, the top pin is ping 3).  Hopefully this helps.

[image]
You can also look at it from the perspective of the L515 camera itself.  With the L515 as in the image above being pointing up on a table top, the left most pin is pin 1, pin 2 is middle and the right most pin is pin 3.

Regarding matching color images with depth, the color images and stream isn't affected by the sync signal.  The color will come normally to the host system.  So you'd have to use something like timestamps or any other metadata to help determine best way of syncing the color images to the depth images you get from each camera at the given timing interval of depth frames in the 8 camera sync setup.

skyek, 2021-01-12:
I am having the identical issue as the creator of this thread. My wires are configured correctly, verified through multimeter probing. 

I have tried various solutions. Switching my signal generator from a Pi to and Arduino, changing from RPI.GPIO to pigpio, changing from RealSense viewer to software, none worked. 

"We also opened all 8 of the L515's depth streams in the RealSense viewer with Intercam Sync Mode and found that our L515's were streaming when the sync signal was LOW not when the sync signal was HIGH, so we switched the two in our code. Is this the intended sync signal on the L515's, where LOW is enabling streaming? If so, it contradicts the image and associated explanation given in the whitepaper in section 2-E:"
Is pretty much exactly what I am experiencing. Firmware is the latest.

skyek, 2021-01-12:
As OP posted, I tried reversing my signals and that also fixed my problem. Thanks OP, but the documentation appears to be wrong.

RealSenseSupport, 2021-01-13:
Thank you for the feedback and we will continue to look over our documentation to improve the information.

closed 2021-01-21 by RealSenseSupport

kkh3049, 2021-01-21:
For what it's worth, I recently checked my wiring and confirmed it to be according to the updated documentation.  Sending a LOW (ground) voltage as my sync signal works while sending a HIGH (3.3V) voltage stops the cameras from collecting depth sensors.