realsenseai/librealsense #7942: launch two cameras at same time

zhhz9706, 2020-12-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D435i | 
| Firmware Version                   | | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.4.0-56-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0  | 
| Language                            |  C++ | 
| Segment			| robot | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi，
How to use a ros node (realsense2_camera) to launch two cameras, for example, mine is realsense t265 & d435i, I want to get odometry and depth at the same time.

ev-mp, 2021-01-05:
@zhhz9706 , please readdress this to [`realsense-ros`](https://github.com/IntelRealSense/realsense-ros) forum

RealSenseSupport, 2021-01-07:
As ev-mp mentioned this can be addressed in the realsense-ros forum

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-13 by RealSenseSupport