realsenseai/librealsense #7944: new issue with last release..

Fred3D-tech, 2020-12-08:
D415 x 3
cat /proc/version
Linux version 5.4.0-56-generic (buildd@lgw01-amd64-025) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020
Using a program which has been working for 2  years, I get after the last update, this error message and the program is not working anymore; 
    profile1 = pipeline1.start(config1)
RuntimeError: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
Intel, are you serious? the program was working fine before.. I spent money to get this program before and it was working. Are you serious? NO YOU ARE NOT. I was ready to order 36 x D415 but when I see how you work. You are just not reliable. I will publish on the social media to prove how you are not reliable. @RealSense-Customer-Engineering it is not good for you. You have a good product but with a really bad support.

Fred3D-tech, 2020-12-08:
we had programs working very well before the last update and they are not working anymore... we were ready to order more realsense cameras and nothing is working anymore... what can we do ? can we trust Intel on that ? We are going to post widely on social medias about that on social medias. Are you reliable and can we invest money on your technology? the answer is NO.

MartyG-RealSense, 2020-12-09:
Hi @Fred3D-tech   If you built SDK 2.40.0 from source code and patched it, would it be possible to build it from source code with the **RSUSB** method instead, as recommended by the 2.40.0 **Release Notes**?

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

The RSUSB method, which is enabled by adding the term **-DFORCE_RSUSB_BACKEND=true** to the CMake build instruction, requires an internet connection to perform the installation but is not dependent on Linux versions or kernel versions and does not require patching.  Instructions for the method can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

closed 2020-12-23 by MartyG-RealSense