realsenseai/librealsense #7945: Support for frame callbacks on MacOS

k2rek, 2020-12-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   |  05.12.09.00  | 
| Operating System & Version |  MacOS (11.0.1)  | 
| Platform                                 |  Macbook Pro  |
| SDK Version                            |  2.40.0  | 
| Language                            |  C++  | 
| Segment			|  others  | 

(Also tested with the same D435 on a different Mac (running OSX 10.14.6, SDK 2.38.1), and found the same results)

### Issue Description
Hi there.  I am working with the SDK on MacOS, and while I am able to successfully fetch frames using `pipe.wait_for_frames()`, I have noticed that callback style of frame fetching (i.e., `pipe.start( [&](rs2::frame f) { someFunction(f); } );`) does not seem to work on this platform.  

In short, 
- simple test programs created using the callback functionality don't work as expected (the callback is never called) 
- the rs-callback example packaged with the SDK produces no output past printing "RealSense callback sample" (expected output being printed frame counters updating once per second)

I will happily expand on the details if it sounds like this is indeed a bug, but first I figured I would check if the callback functionality is actually supported on MacOS?  I couldn't find a definitive answer anywhere as to wether it is or not.

Thanks.

RealSenseSupport, 2020-12-22:
Hi @k2rek 
Sorry I can't reproduce this issue on my two mac system.
Could you share more detail about this issue? or try again?
Could you try to run realsense-viewer first?

My environment is 
MBP (macOS 10.15.7 / 11.1)
Xcode 12.3
RealSense D435 with FW 5.12.9.0 (USB 3.0 Cable)

k2rek, 2020-12-22:
Hi, thanks for the response.

I have since been able to get the callback working in a test program: it was user error on my part related to the scope of the `pipeline` object.  So, that half of the issue is resolved-- sorry for the inconvenience.

However, the rs-callback example included with the SDK still doesn't run properly, as mentioned in the original post.  It's not critical to me, but maybe something you would like to look into regardless, so here are the details:

I'm now on the same environment-- OS 11.1, Xcode 12.3, D435 on 5.12.9.0 connected via USB 3(.2)

I have SDK 2.40.0 installed via Homebrew.

Running realsense-viewer from the installation's folder of examples works as expected, correctly shows USB 3.2 connection and the software is fully functional for streaming data.

Similarly, other included examples such as rs-depth and rs-distance work as expected (displaying correct output in terminal).

However, running rs-callback produces no output beyond "RealSense callback sample":
[image]
(expected output: https://github.com/IntelRealSense/librealsense/tree/master/examples/callback)

RealSenseSupport, 2020-12-24:
It should be an issue about std output stream or terminal.
The example is working at that time.
You can try replace the line `std::cout << "\r";` with `std::cout << std::endl;`
https://github.com/IntelRealSense/librealsense/blob/master/examples/callback/rs-callback.cpp#L60

closed 2021-01-05 by k2rek