realsenseai/librealsense #7946: Calibration needed?

bobdavies2000, 2020-12-08:
[image]

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | latest| 
| Operating System & Version |   Windows 10  | 
| SDK Version                            | latest                        | 
| Language                            |  C#                        | 

I have observed that the depth data has an evenly-spaced linear pattern of zero depth pixels when depth is found for a flat surface.  It is not pronounced in this example so I drew some lines on 3 of the patterns.  The image is straight from the camera - applying the colorizer to the get_depth_frame.  My question: does this pattern of missing depth indicate that my camera calibration is not correct?  I have only 1 D455 camera so I cannot compare it.  The linear pattern is found on any flat surface and will vary in the angle.

MartyG-RealSense, 2020-12-09:
Hi @bobdavies2000   I have not seen an effect exactly like this before and was unable to recreate it.  It does not look obviously like a calibration related issue to me.

If the above image is rendered by your C# application, do you see the same lines on a 2D depth view in the **RealSense Viewer** program, please?

bobdavies2000, 2020-12-09:
The lines do not appear in the imgui interface in the realsense-viewer so thanks for the tip.

closed 2020-12-09 by bobdavies2000