realsenseai/librealsense #7947: If fw.res file required when use the fw.lib

PhoebeHui, 2020-12-09:
Hi All,

The fw.res file required to build the fw.lib, I want to confirm if it's required when use the fw.lib?

I noticed it sets 'PUBLIC' in https://github.com/IntelRealSense/librealsense/blob/master/common/fw/CMakeLists.txt#L46, it would export the target in realsense2Targets.cmake, however, it's not installed, so this would cause usage issue when using the library.

Thank you in advance!

Thanks,
Phoebe

RealSenseSupport, 2020-12-30:
@PhoebeHui Thanks you for pointing out this! This should be a bug. Already forwarded to our engineering team and will let you know if any update. Thanks!

PhoebeHui, 2021-06-09:
Any progress?

RealSenseSupport, 2021-06-09:
@PhoebeHui Sorry, there's no update yet. When get any update will let you know.  Thanks!

ev-mp, 2021-08-23:
@PhoebeHui , the file `fw.lib` is an internal SDK resource used in make's build.
It is not part of the public API and wouldn't be exported.

Did you encounter an issue with the specific target ? Please clarify.

RealSenseSupport, 2021-09-24:
@PhoebeHui Do you still have any questions with the clarification from ev-mp? Looking forward to your update. Thanks!

PhoebeHui, 2021-09-24:
@RealSenseSupport, sorry for late response, I will try to update this port in vcpkg next week, and double check this issue in new release, I will get back to you soon.

PhoebeHui, 2021-09-27:
>@PhoebeHui , the file fw.lib is an internal SDK resource used in make's build.
It is not part of the public API and wouldn't be exported.

Then it should not be set to INTERFACE_LINK_LIBRARIES for target realsense2::fw.

@ev-mp @RealSenseSupport, this problem still exist in latest release revision 2.49.0.

You can repro this issue:

1. Get changes from https://github.com/microsoft/vcpkg/pull/20379
2. ./bootstrap-vcpkg.bat
3. ./vcpkg install realsense2:x64-windows
3. Check ./vcpkg/installed/share/realsense2/realsense2Targets.cmake, it still requires 'common/fw/fw.res' file.
```set_target_properties(realsense2::fw PROPERTIES
  INTERFACE_LINK_LIBRARIES "F:/VCPKG/clean/buildtrees/realsense2/x64-windows-rel/common/fw/fw.res"
```

xaedes, 2022-01-27:
This exact problem still persists (using vcpkgs last version of realsense2 2.50.0).

gilpazintel, 2025-02-03:
@PhoebeHui  / @xaedes can you check on latest version 2.56.
per my check, the path is now relative and not absolute.
thanks
gil

RealSenseCustomerSupport, 2025-02-14:
----------------------------------------------

@PhoebeHui  @xaedes  Did you get chance to try with the latest librealsense v2.56.3 to see if the issue resolved or not?  Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2025-03-04:
----------------------------------------------

@PhoebeHui  @xaedes  We didn't get update from your side for two weeks. As the issue was resolved already, close this ticket accordingly. Please feel free to create another ticket if you still have any issue. Thanks!

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense