realsenseai/librealsense #7949: Question: How to use RS2_FORMAT_MJPEG?

Allius27, 2020-12-09:
Hi there!

SDK has this string - https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_sensor.h#L83

Do d435 or d415 support RS2_FORMAT_MJPEG data format?

I tried to start camera with that argument, but fall in exception

MartyG-RealSense, 2020-12-09:
Hi @Allius27   The SDK release notes for **2.40.0** state that MJPEG is now enabled for **platform** cameras.  This typically means non-RealSense cameras such as webcams.

[image]

The **RealSense Viewer** button has an **Add Source** option at the top of its options side-panel that has a **Platform Camera** option.   Cameras detected as platform cameras will be listed here as selectable.  As a test, when I added my laptop's built-in webcam as the platform camera I was able to select MJPEG from the list of RGB stream modes.

[image]

If a device does not support MJPEG then it gets skipped in the profiles enumeration, according to the link below.

https://github.com/IntelRealSense/librealsense/pull/7693#discussion_r516172538

Allius27, 2020-12-09:
I see, thank. 
In that case d415 doesn't support mjpeg steaming

closed 2020-12-09 by Allius27