realsenseai/librealsense #7950: distance_to_object.ipynb code reads different frames while running different times despite keeping the same range value while pipe.wait_for_frames()?

nirojshrestha019, 2020-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer -->2.38.1.2223) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.40.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  others }                          | 

### Issue Description
I am using this demo code **distance_to_object.ipynb** (https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb)  and running it into my own .bag file that I capture using Intel realsense D435. The issue I am facing is that I get a different frame every time I run it.  Below is the sample line of code and I am using 810 in range value to wait for the 810th frame to capture and save in color_frame and depth frame.

\# Skip 5 first frames to give the Auto-Exposure time to adjust
for x in range(**810**):
&nbsp;&nbsp;&nbsp;&nbsp;pipe.wait_for_frames()
  
\# Store next frameset for later processing:
frameset = pipe.wait_for_frames()
**color_frame** = frameset.get_color_frame()
**depth_frame** = frameset.get_depth_frame()

\# Cleanup:
pipe.stop()
print("Frames Captured")

But it seems that everytime I run the code I am getting different frames in color_frame and depth_frame despite have same range value 810. Can anyone help me how to get same frame everytime I run it?

MartyG-RealSense, 2020-12-09:
Hi @nirojshrestha019   A RealSense Python user who was also skipping frames in a bag took the approach of inserting a **playback.pause()** instruction between pipe.wait_for_frames and the color / depth capture instructions.

https://github.com/IntelRealSense/librealsense/issues/7932

[image]

MartyG-RealSense, 2020-12-23:
Case closed due to no further comments received.

closed 2020-12-23 by MartyG-RealSense

nirojshrestha019, 2022-01-17:
Sorry for the late response. Thank you for your help. It worked.

MartyG-RealSense, 2022-01-17:
No problem at all - it's great to hear that you succeeded.  Thanks very much for the update!