realsenseai/librealsense #7951: D435 Obstacle Detection/Avoidance using Depth Stream

bmils, 2020-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.12.9.0 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC  |
| SDK Version                           |  2.4.0                          | 
| Language                               |  C#                         | 

### Issue Description
I recently took the C# version of the depth console application "cs-tutorial-1-depth," and began to adapt it to my needs. What I am trying to do is have the camera scan the area in front of me and I will stop if an object is in my way. The dilemma I have is that I want to stop when an object is say 2 meters away from me, so I raise the "distance tolerance" of the camera to object to 2 meters. Now the camera at the bottom of the frame (closer up) sees the ground which is approximately 0.7 meters away. I compensate for this by making it so I only stop when objects in the bottom of the frame are say 0.68 meters away, making it so the ground is somewhat ignored. However, this severely reduces the machines ability to stop timely and safely when obstacles are up close. Here's what I have started:
```C#
                    for (int y = 0; y < 480; ++y)
                    {
                        for (int x = 0; x < 640; ++x)
                        {
                            ushort d = depth[x + y * 640];   
                            
                            
                            if (d > 0 && d < (one_meter * 1.65) && y < (fullFrameY * 0.45))
                            {
                                ++coverage[x / 10];
                            }

                            if (d > 0 && d < (one_meter * 1.22) && y < (fullFrameY * 0.8) && y >= (fullFrameY * 0.45))
                            {
                                ++coverage[x / 10];
                            }

                            if (d > 0 && d < (one_meter * 0.8) && y < (fullFrameY * 0.9) && y >= (fullFrameY * 0.8))
                            {
                                ++coverage[x / 10];
                            }

                            if(d > 0 && d < (one_meter * .62) && y >= (fullFrameY * 0.9))
                            {
                                ++coverage[x / 10];
                            }
                        }

                        if (y % 20 == 19)
                        {
                            for (int i = 0; i < coverage.Length; i++)
                            {
                                int c = coverage[i];
                                buffer[b++] = " .:nhBXWW"[c / 25];
                                coverage[i] = 0;
                            }
                            buffer[b++] = '\n';
                        }
                    }
```
Is there a more appropriate use of the depth data for me to accomplish this? I have seen some research articles describe algorithms to "ignore" the floor in the frame and am currently leaning towards creating my own. Any thoughts or comments are appreciated.

MartyG-RealSense, 2020-12-09:
Hi @bmils   Some projects may take a multiple camera approach, such as the **Digit** robot in the link below with 2 RealSense cameras on the front, one on the rear and a camera between its legs for processing floor-based obstacles such as stairs.

https://www.agilityrobotics.com/#our-vision

If you would prefer a single-camera approach, a **RANSAC** algorithm can remove floors with segmentation.  Please google **ransac c#** for more details about this.

Another approach to focusing the image on a particular area of the image whilst isolating other areas (e.g the lower half of the image) is to define a **bounding box** for X and Y that excludes all data outside of the box's boundaries.

The easiest method though may be to use the RealSense SDK's **High Speed Capture** mode.  This mode, which is compatible with D435, allows you to run at 300 FPS with an 848x100 field of view focused on the center-line of the screen.  This could make it practical to detect objects at close range whilst keeping the focus away from the floor at close-range.  Because of the reduced field of view, it can run at 300 FPS whilst only having processing demands comparable to 90 FPS full-screen.

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

bmils, 2020-12-14:
I appreciate the quick response and suggestions. If no one else comments or has another approach I can take, feel free to close this issue in a few days @MartyG-RealSense.

MartyG-RealSense, 2020-12-14:
Thanks very much @bmils  - good luck!

closed 2020-12-21 by MartyG-RealSense