realsenseai/librealsense #7952: Realsense Viewer not detecting L515 (Win10, Surface Pro 3)

PhinPhin, 2020-12-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L151 } | 
| Firmware Version                   | no idea... | 
| Operating System & Version | Win 10  |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.40                          | 
| Language                            |  ?                          | 
| Segment			|  3D-scanning }                          | 

### Issue Description

Realsense Viewer won't detect the camera. Right now the camera is detected by Windows and shown in the devicemanager but not in Realsense Viewer. Unplugging won't help...

[image]

I also tried to run the firmware-updatetool but I get this error (sorry, german):

"Intel.RealSense.FW.Update.exe -l
 09/12 17:57:58,540 ERROR [8632] (types.h:313) CoCreateInstance(CLSID_SensorManager, NULL, CLSCTX_INPROC_SERVER, IID_PPV_ARGS(&pSensorManager)) returned: HResult 0x800704ec: "Dieses Programm wurde durch eine Gruppenrichtlinie geblockt. Wenden Sie sich an den Systemadministrator, um weitere Informationen zu erhalten."
 09/12 17:57:58,580 ERROR [8632] (types.h:313) CoCreateInstance(CLSID_SensorManager, NULL, CLSCTX_INPROC_SERVER, IID_PPV_ARGS(&pSensorManager)) returned: HResult 0x800704ec: "Dieses Programm wurde durch eine Gruppenrichtlinie geblockt. Wenden Sie sich an den Systemadministrator, um weitere Informationen zu erhalten.""

I tried to allow non-Store apps but nothing so far.... help?

RealSenseSupport, 2020-12-09:
Hi @PhinPhin 

Couple questions or things I'm hoping you can try in order for us to determine what might be going wrong:

- When did you receive this camera?  Was it recent?  Have you had the camera working before trying Viewer from SDK 2.40?
- Have you changed out the USB cable?
- Is there anyway you can go back to an older Viewer version from both the 2.38.1 (https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1) and 2.39 (https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0).  Once you do this let me know if any of these Viewers can see the L515 camera.

PhinPhin, 2020-12-10:
Hey, 

I received the camera yesterday, no I just installed the latest version of the SDK, as your "getting started" guide told me.
I tried a different USB-C cable, no luck. But as I said, Windows detects it.
Sure, I can try to install a older version of the SDK. I will report back once I tried.

ofl1, 2020-12-13:
Hi @PhinPhin

What is the name of the USB host controller that controls your USB3 port? Which driver is installed for this?

I am asking this because I have a similar problem and I want to find the cause.

PhinPhin, 2020-12-14:
8086:9C31

Intel(R) USB 3.0 eXtensible-Hostcontroller – 1.0 (Microsoft)

It's a Surface Pro 3....

RealSenseSupport, 2020-12-15:
Hi @PhinPhin and @ofl1 

Thanks for information. We'd like to discuss with you directly. We have some questions regarding serial number, maybe some personal questions, etc.

Can you submit a ticket at the following location? http://realsensesupport.intel.com/

PhinPhin, 2020-12-24:
> Hi @PhinPhin and @ofl1
> 
> Thanks for information. We'd like to discuss with you directly. We have some questions regarding serial number, maybe some personal questions, etc.
> 
> Can you submit a ticket at the following location? http://realsensesupport.intel.com/

Well, I guess that is a joke, right? I did as you told me and I get this reply:

[image]

tedtman, 2020-12-30:
Can the Win10 camera app run the RGB imager? What are the driver details in device manager for each imager and the composite device?

ev-mp, 2021-01-05:
@PhinPhin , L515 make sure the device is recognized as USB3 using Win-dedicated tools as `USBView` to check the BCD string.
Support for USB2 was provided with the release of FW version 1.5.1.3 (October 20`), so if you hadn't upgraded a FW it could definitely be the reason. 
Possible duplicate of #7224, #7442

ofl1, 2021-01-06:
> 8086:9C31
> 
> Intel(R) USB 3.0 eXtensible-Hostcontroller – 1.0 (Microsoft)
> 
> It's a Surface Pro 3....

@PhinPhin , thanks for specifying the USB controller. There's another one (ASMedia) in my PCs. It's probably not the problem.

> @PhinPhin , make sure the device is recognized as USB3 using Win-dedicated tools as `USBView` to check the BCD string.
> Support for USB2 was provided with the release of FW version 1.5.1.3 (October 20`), so if you hadn't upgraded a FW it could definitely be the reason.
> Possible duplicate of #7224, #7442

@ev-mp , thanks for the tip. With `USBView` I can now see that my L515 is only recognized as a USB2 device under Windows too. I've tried everything (see #7843).
Under these circumstances the FW upgrade could be difficult. If that somehow succeeds the question is: Is the full range of functions available in USB2 mode? Or are there any restrictions then?

ev-mp, 2021-01-07:
@PhinPhin , can you verify that camera access is enabled in Windows Privacy settings ? (#3381)
I would also recommend to put self-powered USB3 hub between Surface and Camera - in the past we've encountered cases  especially with portable devices struggling to provide adequate power for Device to complete USB3 enumeration successfully.

PhinPhin, 2021-01-10:
> @PhinPhin , can you verify that camera access is enabled in Windows Privacy settings ? (#3381)
> I would also recommend to put self-powered USB3 hub between Surface and Camera - in the past we've encountered cases especially with portable/mini computers struggling to provide adequate power for Device to complete USB3 enumeration successfully.
> 
> As for USB2 vs USB3 - the difference is in the supported streaming resolutions and FPS rates which are bounded by USB2 bandwidth and lower maximal power.
> The actual numbers may vary between FW versions, thus you can use `rs-enumerate-devices` tool to check for the current configuration.

Hi, sorry for the late reply, work was a bit demanding the last 2 weeks. I found another Notebook with USB3 at work that could use the camera in the RealSense Viewer, too. I could update the firmware with it, I also used the most recent Github-download to update the firmware, too. 

USB Devicetreeview shows me this:

`
    =========================== USB Port10 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-10
Properties               : 0x03
 IsUserConnectable       : yes
 PortIsDebugCapable      : yes
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x0A (Port 10)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#4&e34fdde&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x01 (Port 1)
 -> CompanionPortChain   : 1-1

      ========================== Summary =========================
Vendor ID                : 0x8086 (Intel Corporation)
Product ID               : 0x0B64
USB version              : 3.2
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self Powered             : yes
Demanded Current         : 440 mA
Used Endpoints           : 9

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB-Verbundgerät
Device Path              : \\?\USB#VID_8086&PID_0B64#00000000F0270221#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-6
Device ID                : USB\VID_8086&PID_0B64\00000000F0270221
Hardware IDs             : USB\VID_8086&PID_0B64&REV_1052 USB\VID_8086&PID_0B64
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0005 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.19041.488  Date: 2020-11-14)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0010.Hub_#0001
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(10), ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(SSP1)
Container ID             : {6bf8851a-77bb-5d11-a7da-1ca3350bc5ac}
Manufacturer Info        : (Standard-USB-Hostcontroller)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 10
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : Intel(R) RealSense(TM) 515 Depth (USB-Videogerät)
  Device Path            : \\?\USB#VID_8086&PID_0B64&MI_00#6&959bca1&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_CAPTURE)
[... 80 lines cut ...]
 ReservedMBZ             : 0x00
Data (HexDump)           : 0A 00 00 00 10 00 00 00 04 00 00 00 03 00 00 00   ................

    ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x320 (USB Version 3.20)
bDeviceClass             : 0xEF (Miscellaneous)
bDeviceSubClass          : 0x02
bDeviceProtocol          : 0x01 (IAD - Interface Association Descriptor)
bMaxPacketSize0          : 0x09 (9 bytes)
idVendor                 : 0x8086 (Intel Corporation)
idProduct                : 0x0B64
bcdDevice                : 0x1052
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "Intel(R) RealSense(TM) Camera"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "Intel(R) RealSense(TM) 515"
iSerialNumber            : 0x03 (String Descriptor 3)
 Language 0x0409         : "00000000F0270221"
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 20 03 EF 02 01 09 86 80 64 0B 52 10 01 02   .. .......d.R...
                           03 01                                           `

So, it is detected as USB3?

ev-mp, 2021-01-10:
@PhinPhin , using the alternative laptop the L515 is recognized and enumerated properly as USB3 device:

>Vendor ID : 0x8086 (Intel Corporation)
Product ID : 0x0B64
USB version : 3.2

>bcdUSB : 0x320 (USB Version 3.20)

Additionally you've managed to flash new FW.
This allows to rule out Camera/Cable issues.

So you need to look into Surface's HW and Software configuration, such as USB Power and Windows permissions as mentioned above.

PhinPhin, 2021-01-10:
The only way to test the USB-port for power-issues are a few USB2 and USB3-Hardisks, all work at full bandwith. No power-issue.

I have no clue why a group-policy could be enabled.... it is my own tablet. Any way to list up all policies in use? I checked and the camera-access is not restricted. The RealSenseViewer is not shown in the applist for the camera-access, though.

RealSenseSupport, 2021-01-13:
Have you tried a powered USB Hub?  Something like this:  https://www.amazon.com/dp/B00DQFGH80/ref=asc_df_B00DQFGH805315309/?tag=hyprod-20&creative=394997&creativeASIN=B00DQFGH80&linkCode=df0&hvadid=167151358503&hvpos=1o1&hvnetw=g&hvrand=9308811005364068537&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032905&hvtargid=pla-281221827342

Not familiar with any system level policies on the Surface Pro 3 that could get in the way.

PhinPhin, 2021-01-24:
Sorry, corona-lockdown slowed down everything here. 

I got a powered USB3-hub just to test it, and as it was before: Devicemanager lists RGB and Depthcam but the RealSenseViewer won't show the camera. While it starts is shows the message that it detected a RealSense device, though.

Just saying, in one of your own videos a Surface Pro is used: https://www.youtube.com/watch?v=41Yu2H9_z3w

PhinPhin, 2021-01-24:
Oh and my support-ticket on the RealSense support website vanished without further reply.... great.

shuixing85, 2021-02-04:
Hi there, 
have you tried to disable the other camera?

PhinPhin, 2021-02-04:
Define disabled? In devicemanager? Not yet. But why would that help?

RealSenseSupport, 2021-03-03:
Hi @PhinPhin 

Can you let us know where you are with your L515 at the moment?  Did you submit RMA for the product?

closed 2021-03-23 by RealSenseSupport