realsenseai/librealsense #7955: Is get_timestamp() aware of serial communication latency?

amm385, 2020-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | don't have it on me :( | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  2.36.0                         | 
| Language                            |  python                         | 
| Segment			|  Robot                         | 

### Issue Description
Imagine I have a very long usb cable or a very unpredictable latency between my image being ready on the camera firmware and my pyrealsense sdk receiving the data. If I call get_timestamp(), I can imagine there being one of two values returned, depending on how get_timestamp syncs its clock with my system clock:

a) My host computer's timestamp **when the frames were fully collected on the camera's firmware**
b) My host computer's timestamp **when the frames were received by the host computer**

Which of those values is returned by get_timestamp?

If this doesn't make sense yet, here's an example:
The camera and my computer are clock-synced. The camera takes a picture at t=100ms.
It transmits this information back to the computer. This process takes 10ms.
So the SDK receives the full batch of data at t=110ms.

Does get_timestamp() return 100 or 110?

Or, alternatively, am I misunderstanding how this works? I did my best to parse the other questions on here but am still confused.

MartyG-RealSense, 2020-12-10:
Hi @amm385   The documentation for **get_timestamp()** in the link below states that it retrieves the time at which the frame was captured.  At runtime the SDK selects the most correct representation from the different types of timestamp available, based on both device and host capabilities.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

The timestamp timings of RealSense camera sensors are generated as frame metadata attributes.  These can be produced by the camera firmware or by the host clock.  The links below describe how metadata is created.

https://dev.intelrealsense.com/docs/frame-metadata 

https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 

Device time and system time are aligned when Global Time is enabled.

https://github.com/IntelRealSense/librealsense/pull/3909 

https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-553590316

amm385, 2020-12-10:
Thanks for your answer @MartyG-RealSense. Certainly a bit confusing but glad there are a variety of options to choose from.

closed 2020-12-10 by amm385

MartyG-RealSense, 2020-12-11:
You are very welcome @amm385 - good luck!