realsenseai/librealsense #7957: Emitter Issues with D435I

ykokit, 2020-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435I | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version | Linux (Ubuntu 18.04.5 LTS) / Windows 10  | 
| Kernel Version (Linux Only)    |  5.3.0-62-generic | 
| Platform                                 | PC |
| SDK Version                            | legacy / 2...                        | 
| Language                            | python                         | 

### Issue Description
When using D435 to capture infrared data, Infared 1/Y8 is turned on, and Emitter On/Off is enabled. However, when extracting the .bag file (which is then converted to a collection of .png files), the results of the data files vary:
   * Data is collected as expected
   * Data looks as if the emitter is always on, resulting in the collected images having dots over the face
   * Data has interchanging output where there will be expected and dotted results in the same file (most common)
    * Data has interchanging output where there will be results with larger/brighter and smaller/dimmer dots (occurs in dim lighting, avoided by turning off auto exposure for infrared camera)

This has been attempted on both a Linux and Windows os with the same results. Has anyone else come across the same issues with data collection?

Thank you

MartyG-RealSense, 2020-12-09:
Hi @ykokit    When you use the emitter on / off mode, does the dot pattern appear on frames where the pattern is meant to be "off"?

If so, does the behavior of the patterns change during the day depending on the current lighting level?  For example, in good lighting conditions the pattern might be visible on some types of material but not others.  And then later in the day when the light level may be dimmer, the behaviors seem reversed - the dots are visible on the materials they were previously invisible on, and invisible on the materials they were earlier visible on?

If the above behavior resembles the results you are experiencing then there may not be much that can be done in terms of a software solution.  Instead, if you need to use the emitter on / off mode then I would recommend using the camera in lighting conditions that are as stable and consistent as possible during the day.

ykokit, 2020-12-11:
Yes, when using the emitter on / off mode, does the dot pattern appear on frames where the pattern is meant to be "off". And no, while I understand what you mean in regard to the dots showing up on some materials and not others (I have extracted data that contains the behavior you are referring to) I have not come across it being reversed, because the dots would be visible on all materials. The behavior is effected by lighting to some degree (after a certain dim-ness this certain issue is hard to avoid), but I would make multiple .bag files under good lighting conditions and would still get varying results working with the same lighting. 

The reason why is issue is being brought up is because we collected data of around 20 people without this problem arising before, and there is some confusion as to why it has arisen now. Could there be any other possibilities as to what it may be?

MartyG-RealSense, 2020-12-12:
I read your case again from the beginning.  It sounds as though you are only treaming infrared and not depth, and that you do not want the IR dots appearing on the image.

If that is correct, then a simple solution may be to disable the IR emitter.  This will remove the dots from the IR pattern but preerve the IR image, as the IR Emitter and the left / right IR sensors are separate components.

Disabling the IR Emitter can have a significant impact on the depth image.  However, if you are only streaming IR then it should be fine to completely turn off the IR Emitter to remove the dot pattern from all frames.

The emitter can be disabled in Python using scripting in the link below:

https://github.com/IntelRealSense/librealsense/issues/7919#issuecomment-738857679

You can also disable the IR Emitter in the **RealSense Viewer** by setting the drop-down menu of the **Emitter Enabled** option to **Off** (the default On-state is 'Laser').  This is different from the **Emitter On Off** option further down the options side-panel, which causes the dot pattern to rapidly alternate on and off automatically and is not an on-off button for the emitter.

[image]

[image]

[image]

closed 2020-12-26 by MartyG-RealSense