realsenseai/librealsense #7959: Convert software_video_frame object to frameset

DragonDeux, 2020-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

DragonDeux, 2020-12-09:
Hello Realsense engineers, 

Could you please help to explain how to convert software_video_frame object to frameset, then align depth frame to this software_video_frame.

part of code: (i have only enable depth stream on the pipeline)
```
    vid_frame = rs.software_video_frame()
    vid_frame.stride = 1920*2
    vid_frame.bpp = 2
    vid_frame.timestamp = 0.0
    vid_frame.pixels = convertedImage
    vid_frame.domain = rs.timestamp_domain.hardware_clock
    vid_frame.frame_number = 0
    vid_frame.profile = colorProfile
    sensor.on_video_frame(vid_frame)
    Depth = pipeline.wait_for_frames()
    aligned_frames = align.process(vid_frame)
```
for exemple.

Thanks in advance for helping

DragonDeux, 2020-12-09:
i'm trying to using intel realsense Lidar incorporate with a non-realsense RGB camera, the aim is, to align a Lidar depth to non-realsense RGB color image with a calibrated parameters(intrinsic and extrinsic, these i already have). i'm trying integrated non-realsense RGB camera image to Realsense SDK frameset as i attached code in above Q/A, but there is no sample to give some helps in this topic, such as how to convert a soft synthetic image(from non-realsense device) into Realsense frameset even without synchronization requirement.

ev-mp, 2021-01-05:
@DragonDeux hello, did you check [`rs-software-device`](https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device) demo - it closely resembles the described scenario

RealSenseSupport, 2021-01-13:
Hi @DragonDeux 

Were you able to try ev-mp suggestion?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-29 by RealSenseSupport