realsenseai/librealsense #7965: Realsense viewer showing data only after hardware reset or reboot PC

aidonchuk, 2020-12-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-54-generic                                         | 
| Platform                                 | amd64  |
| SDK Version                            |   2.40.0                          | 
| Language                            |  python                          | 
| Segment			|  others                           | 

Hi! I'm using [this](https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842) to install from sources and everything installs properly except python =) 
But main problem that i have to do a hardware restart each time i want to see the camera output in realsence-viewer. 

I'm using stock usb cable(witch i tested in Windows and everything is ok).
After hard reset or reboot camera works properly
[image]

When I turn off and turn on again Depth Sensor
[image]

I cant see data on screen and only update of warnings added 
[image]

> Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0265552
    Firmware Version              : 	01.05.02.00
    Recommended Firmware Version  : 	01.05.02.00
    Physical Port                 : 	2-3-4
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	0003ac809dc1
    Firmware Update Id            : 	0003ac809dc1

Pls help!

aidonchuk, 2020-12-10:
Update about python!
Like with realsense-viewer on first execute after reboot camera works.
But second time I get this 
[image]

Additionally after hard reset python still don't get responce.

ps I'm absolutely stuck with my development =((

aidonchuk, 2020-12-13:
@RealSenseSupport Hi! Can you help me? please

RealSenseSupport, 2021-01-12:
Hi @aidonchuk 

What type of platform are you using the L515 on?  Is it a single L515 camera that you are using or are you trying to use multiple L515 cameras at the same time?

Are you using any powered USB hubs in your setup or are you connected directly to the system USB3 port?  What USB controller are you plugged into?

closed 2021-03-16 by RealSenseSupport