realsenseai/librealsense #7966: calibration through water

Cimino023, 2020-12-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer ) | 
| Operating System & Version |   {Ubuntu 18.04}  | 
| Kernel Version (Linux Only)    |  (4.9.140)                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Other}                          | 

### Issue Description
<
Hi! I would like to use my D435i to calculate dimensions of objects inside an aquarium.
The camera is placed just behind the aquarium glass (which is of known thickness) and the tank is big enough to assume that the water body continues indefinitely after the glass.
I would like to be able to calibrate the camera without disturbing trigonometry and light refraction angles, but rather using graphical methods such as submerging a ruler in the aquarium and manually indicating real distances between pixels as shown by the ruler itself. Or maybe by having a reference point in the aquarium and sliding the camera on the aquarium glass by some known distance. Or maybe even with the on-chip calibration tool? I don't think I can manage to do this easily myself, could you please help me?
>

MartyG-RealSense, 2020-12-10:
Hi @Cimino023   The stereo depth algorithms of the 400 Series cameras work excellently underwater, with the camera in a waterproof container.  They can also depth-sense through the glass of an aquarium from outside of it.  Intel have published a Python tutorial called **depth_under_water** for such sensing.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb

As just one practical example of how RealSene has been used for underwater projects, the YouTube link below highlights a project called **FishSense** for measuring the size and population of fishes in an ocean-size environment.

https://www.youtube.com/watch?v=50Y1NiyX1D0

In regard to calibration: assuming that you cannot insert an object that could be degraded or destroyed by contact with water, I can see a couple of possibilities:

-  Insert a flat, low-texture surface in the aquarium (like the side-wall of an aquarium castle decoration) that is large enough to fill the camera's view and use it as a target for the On-Chip calibration.

-  Use the **Dynamic Calibrator** tool and hold a calibration target up to the outside of the glass, with the camera observing the target through the glass from within the aquarium.  The target can be a paper print-out mounted on a solid surface, or a digital target displayed on the screen of an Android device.

Cimino023, 2020-12-11:
> Hi @Cimino023 The stereo depth algorithms of the 400 Series cameras work excellently underwater, with the camera in a waterproof container. They can also depth-sense through the glass of an aquarium from outside of it. Intel have published a Python tutorial called **depth_under_water** for such sensing.
> 
> https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb
> 
> As just one practical example of how RealSene has been used for underwater projects, the YouTube link below highlights a project called **Fish Sense** for measuring the size and population of fishes in an ocean-size environment.
> 
> https://www.youtube.com/watch?v=50Y1NiyX1D0
> 
> In regard to calibration: assuming that you cannot insert an object that could be degraded or destroyed by contact with water, I can think of a couple of initial possibilities:
> 
> * Insert a flat, low-texture surface in the aquarium (like the side-wall of an aquarium castle decoration) that is large enough to fill the camera's view and use it as a target for the On-Chip calibration.
> * Use the **Dynamic Calibrator** tool and hold a calibration target up to the outside of the glass, with the camera observing the target through the glass from within the aquarium.  The target can be a paper print-out mounted on a solid surface, or a digital target displayed on the screen of an Android device.

WOW! This is great, thank you!

closed 2020-12-24 by MartyG-RealSense