realsenseai/librealsense #7969: compiling on ubuntu 20.04 requires an opengl cmake change

erh, 2020-12-10:
Because of the dependencies on ubuntu, we need to specify to use the modern opengl libraries.
I think this change is safe everywhere, and certainly worked for me.

```
diff --git a/CMake/opengl_config.cmake b/CMake/opengl_config.cmake
index 3c75caf03..047c705b6 100644
--- a/CMake/opengl_config.cmake
+++ b/CMake/opengl_config.cmake
@@ -1,2 +1,3 @@
+set(OpenGL_GL_PREFERENCE "GLVND")
 find_package(OpenGL REQUIRED)
 set(DEPENDENCIES realsense2 glfw ${OPENGL_LIBRARIES})

```

MartyG-RealSense, 2020-12-11:
Thanks very much @erh for your advice.  It should be a useful reference if librealsense users with Ubuntu 20 04 have problems with OpenGL in future.

MartyG-RealSense, 2020-12-20:
Hi @erh   Do you require further assistance with this case, please?   Thanks!

erh, 2020-12-20:
No, all set, thanks.
I'll just keep applying that patch locally  :)

closed 2020-12-20 by erh