realsenseai/librealsense #7972: frame metadata backend_timestamp = 0?

amm385, 2020-12-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | don't have it on me :( | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  2.36.0                         | 
| Language                            |  python                         | 
| Segment			|  Robot                         | 

### Issue Description
I'm getting a value of 0 for my frame's backend_timestamp value:
`frame.get_frame_metadata(rs.frame_metadata_value.backend_timestamp)`
`frame.get_frame_metadata(rs.frame_metadata_value.time_of_arrival)` and `frame.get_timestamp()` are both working correctly. **What does this mean?** From what I've read on this forum, `sensor_timestamp` and `frame_timestamp` require a kernel patch, but `time_of_arrival` should be supported universally.

MartyG-RealSense, 2020-12-11:
Hi @amm385   I extensively researched your issue with backend_timestamp being zero.  My understanding is that backend_timestamp is a **system time** timestamp that is dependent on the OS, and not a firmware-generated timestamp.  So there was no clear reason why the timestamp would be 0.

My research did find a case though where a couple of RealSense users encountered this zero result on Raspberry Pi, which like Jetson uses an Arm processor.  There was also a case of a zero result for this timestamp on an Android device, which typically use Arm processors.  So I wonder whether the zero result is Arm-related.

amm385, 2020-12-11:
Thanks for the response, I too was suspecting it might be related to the ARM architecture, or the kernel build specific to that architecture.

Interestingly, I have a separate arm computer that is working fine, so it must be something very specific to the instance i'm having issues with. (The computer I have that's functioning is a [Jetson TX2](https://developer.nvidia.com/embedded/jetson-tx2-developer-kit), the one that's returning 0 is an [industrialized TX2](https://www.advantech.com/products/9140b94e-bcfa-4aa4-8df2-1145026ad613/mic-720ai/mod_19d7f198-a3f3-4975-ac87-e8facd1045b3))

Either way, good to know this is not just a simple user error on my end, will continue to look into it

MartyG-RealSense, 2020-12-11:
Thanks very much @amm385 for the update.  Good luck with your investigations!

MartyG-RealSense, 2020-12-18:
Hi @amm385  Do you require further assistance with this case, please?   Thanks!

amm385, 2020-12-19:
tbh I've put this problem on the backburner for now. If I revisit and figure something out, I'll be sure to update, but no assistance needed for now

MartyG-RealSense, 2020-12-19:
Okay, I will close the case for now.  Please feel free to create a new case if you return to the problem and require assistance.  Good luck, and thanks for the update!

closed 2020-12-19 by MartyG-RealSense