realsenseai/librealsense #7975: Computing power needed

oscarthorn, 2020-12-11:
Hi!

What is the cost in terms of processing power to get depth data from the D455? In terms of CPU? In terms of GPU (if needed)?

We are currently working on a embedded system (based on Xavier NX) and so we don't have much computing power to spare. Currently we are using the ZED2 but since the D455 seems to have some internal processing power we are wondering if it would reduce the load on the host compared to the ZED?

MartyG-RealSense, 2020-12-11:
Hi @oscarthorn   The 400 Series cameras have a built-in **Vision Processor D4** board that enables the camera to do processing work on captured camera data before sending it to the computer instead of doing all the work on the computer's GPU.  This enables the 400 Series to run on low-specification, low power computing equipment.  

The company Stemmer Imaging have a document on their site that provides more details about the D4.

https://www.stemmer-imaging.com/media/uploads/cameras/13/136603-Intel-RealSense-D4-Vision-Processor-Brochure.pdf

oscarthorn, 2020-12-11:
Thanks for the quick answer, seems promising!

agrunnet, 2020-12-11:
@oscarthorn yes I want to second what Marty said. The answer is “basically zero load on host CPU” because all depth processing happens on embedded ASIC. We have examples of full applications that run to read depth at 848x480 at 90fps, and track finger locations, for example, with CPU load for whole app being less than 5%. No GPU required.

That being said of course you can add processing to do more things, such as extra post processing.

MartyG-RealSense, 2020-12-19:
Hi @oscarthorn   Do you require further assistance with this case, please?   Thanks!

oscarthorn, 2020-12-21:
No, thanks for the help!

closed 2020-12-21 by oscarthorn

MartyG-RealSense, 2020-12-21:
Thanks very much @oscarthorn for the update   :)