realsenseai/librealsense #7976: SDK V2.4.40 - testing and problems on different OS on the same NUC (Ubuntu 18.04 - 20.04) Windows 10 + Jetson AGX Xavier

Fred3D-tech, 2020-12-11:
computer: Intel Nuc Kit NUC8i3BEH /Ram 16Gb / SSd 1 Tb
Ubuntu 20.04 
Ubuntu 18.04 
Windows 10 pro

Jetson AGX Xavier
Ubuntu 18.04

Intel® RealSense™ SDK 2.0 (v2.40.0)
All the D415 have the last firmware (05.12.09.00)

I tested everytime a build from source and the dkms or exe available. I also used the anaconda packages or pip packages.
The results are the same.

I spend my week testing different OS on the same platform with the last SDK. For my tests I used a piece of code (python) which was working well until september.
I also decided to test the same code on a Jetson AGX Xavier.
It was very interesting, even if it took me a lot of time.
The Ubuntu install is using a fresh install and using a sh script I wrote to install realsense sdk and open3d.

same python 3.7 code with 3x D415.
realsense-viewer is working well on all the platforms even with the 3 cameras and it is possible to access advanced mode.
Realsense viewer with the 3 cameras activated give 40-60 Fps
tested with a direct connection to usb 3 ports and also with with powered USB 3 hub (AmazonBasics Hub USB-C 3.1 7 ports power 36 W (12 V/3 A)

On Ubuntu 18.04 and Ubuntu 20.04 : 
after the activation of the first camera (sometimes the second), you get this message in the python script :
RuntimeError: Frame didn't arrive within 5000
is an exception thrown at frames = pipeline.wait_for_frames()
It looks like on Ubuntu 20.04, it happens sometimes with the activation of the first camera and with Ubuntu 18.04, it happens more with the activation of the second camera. 
In all cases, you can't activate the 2 cameras (and not the 3 of course)

On Jetson AGX Xavier:
Error message: failed to set power state (even using the powered hub) when trying to access advanced mode in python. Then the code is not working at all.

With the Nuc with Windows 10 pro:
Same script, no problem using 3x D415 connected. It works well everytime with the 3 cameras.

Does it mean that only the Windows 10 version of the sdk is stable?
I'm not a fan of windows 10. I prefer Ubuntu. But I guess we have no choice.

MartyG-RealSense, 2020-12-12:
Hi @Fred3D-tech   Issues with multi-camera setups, such as **Failed to set power state** when using more than one camera, have been known to occur on some systems.  Improvements to how rs2::pipeline is handled in multi-cam setups were introduced in SDK **2.35.2**, though these enhancements may not solve problems that are related to causes other than rs2::pipeline, such as an issue with a particular model of USB hub.

In a case about multi-cam problems in the link below, a RealSense user offered a script that worked for their particular problem.

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-625671835

aidonchuk, 2020-12-12:
Same for me https://github.com/IntelRealSense/librealsense/issues/7965
but only after first connect-disconnect after reboot. 
And on windows is ok. Me too use ubuntu only.

Fred3D-tech, 2020-12-18:
what I don't understand is why using the same python script on the same machine, on Windows 10, I get excepted results with 3 x d415.. On Ubuntu it works with only with 1 (and sometimes 2) cameras.. 
On the Jetson Xavier, I tested everything and even with only one camera, I get this message  Error message: failed to set power state.
I changed the code but I still have the same problem on the jetson Xavier, trying to enter advanced mode.
I think you have a very big problem.

MartyG-RealSense, 2020-12-19:
Hi @Fred3D-tech   I considered all of the information posted on this case very carefully.   Thakns very much for the detailed feedback.

If you are using a multi-camera setup then building from DKMS packages or a manual source code build of librealsense with kernel patching will be best suited to your situation.   This is because an **RSUSB backend** build is suited to single-camera setups (you do not state that you are using RSUSB though).

If your setup worked up until September, it may be worth testing with SDK **2.38.1** and firmware **5.12.7.100** on your Linux machines, as this marks the point before internal changes to the firmware from 5.12.8.200 onwards meant that recent SDK versions like 2.39.0 and 2.40.0 should be used.

Having said that, there have been reported issues with **Failed to set power state** on Jetson boards (Nano, TX2, Xavier) on multi-cam setups prior to September 2020.  SDK **2.35.2** introduced multicam improvements, though they address problems related specifically to rs2::pipeline rather than other factors that might cause problems (such as a particular brand of USB hub).  So it is not necessarily a problem that is related directly to librealsense.

Fred3D-tech, 2020-12-21:
> Having said that, there have been reported issues prior to September 2020 with **Failed to set power state** on diferent types of Jetson board (Nano, TX2 and Xavier) on multi-cam setups of more than one camera.

on the Jetson Xavier, I have the problem Error message: failed to set power state with just one camera now !

MartyG-RealSense, 2020-12-22:
Some RealSense users have found that they achieve great stability, particularly in multicam setups, if they add a time delay after the pipeline start.  An example of this is a **500 ms** delay used in the link below to resolve a **failed to set power state** error when starting one or more cameras:

https://github.com/IntelRealSense/librealsense/issues/6120#issuecomment-704910981

MartyG-RealSense, 2021-01-06:
Case closed due to no further comments received.

closed 2021-01-06 by MartyG-RealSense

funshine, 2021-01-06:
same "failed to set power state" problem here, only one L515 with Jetson NX,
compiled Open3D with librealsense,
I have tried set usbcore.autosuspend=-1, no help
with "dmesg --follow", when I unplug/plug L515, shows:
uvcvideo: Unable to create debugfs 2-5 directory.
...

funshine, 2021-01-09:
After install librealsense follow [installation_jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) 
`sudo apt-get install librealsense2-utils`
`sudo apt-get install librealsense2-dev`
the realsense-viewer works great, and the previous compiled app also works. 
Thank you

MartyG-RealSense, 2021-01-09:
Thanks very much @funshine for the update about your success!