realsenseai/librealsense #7982: File size of library when building from source is very large

davidcann, 2020-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | any | 
| Operating System & Version |   Linux (Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.9.140                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.40.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
When building the library from source, the file size of librealsense2.so.2.xx.x ends up being around 200 MB, but the librealsense2.so.2.xx.x installed via the "realsense-hw-public" apt repository is around 19 MB (most recent is 2.35.2 for arm64 on Nvidia Jetson). We need 2.40.0, but I've confirmed that the issue is the same with 2.35.2.

Build command used:

    cmake .. -DBUILD_EXAMPLES=false -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j4

I realize that there are several packages in the apt repository, but we want to use the minimal installation required on our client devices. Is there a way for us to build a minimal library that is smaller than the ~200 MB .so file?

MartyG-RealSense, 2020-12-13:
Hi @davidcann   Another RealSense user encountered the same issue in the past with package versus source build size.  A RealSense team member advised them to strip debug symbols out of the **.so** file.

https://github.com/IntelRealSense/librealsense/issues/3211

davidcann, 2020-12-13:
Thanks, that worked.

closed 2020-12-13 by davidcann