realsenseai/librealsense #7986: Firmware update fail D415

wodndb, 2020-12-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.09.13.00 | 
| Operating System & Version |  Win10 Pro 20H2  | 
| Platform                                 | PC |
| SDK Version                            | 2.39.0 | 
| Language                            | C#, Unity  | 

### Issue Description
I tried to update firmware of my D415 camera using RealSense Viewer, but I failed.
```
UNKNOWN in rs2_create_device(info_list:00000217AE13D200, index:0):
Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
```

So I tried to use firmware update tool in RealSense SDK, but it also not work.

```
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-fw-update.exe -l

connected devices:
1) Name: Intel RealSense D415, serial number: 836212060725, update serial number: 837313020585, firmware version: 05.09.13.00, USB type: 3.2

C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-fw-update.exe -s 836212060725 -f Signed_Image_UVC_5_12_9_0.bin

search for device with serial number: 836212060725

updating device:
Name: Intel RealSense D415, serial number: 836212060725, update serial number: 837313020585, firmware version: 05.09.13.00, USB type: 3.2
 14/12 13:31:53,962 ERROR [5048] (types.h:313) hr returned: HResult 0x8007001f: "?
```

How to update D415 firmware in this case?

MartyG-RealSense, 2020-12-14:
Hi @wodndb 

Could you use th einstructions in the link below to check in the Windows privacy settings that Windows is allowed to access the camera hardware and camera applications, please?

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

wodndb, 2020-12-14:
Hi @MartyG-RealSense 

Thanks to reply this issue.

Unfortunately my computer already allowed to access camera.

[image]

[image]

MartyG-RealSense, 2020-12-14:
Thank you for checking,  Next, may I ask whether you are updating the firmware using the official 1 meter USB cable supplied with the camera, or with a longer USB cable of your own choice?

wodndb, 2020-12-14:
@MartyG-RealSense I tried to both cables.

MartyG-RealSense, 2020-12-14:
In SDK 2.39.0, when you used the RealSense Viewer did you select the **Install Recommended Firmware** option from the **More** option at the top of the Viewer's options side-panel, please?

This should install firmware **5.12.8.200** (the recommended firmware for SDK 2.39.0).

[image]

wodndb, 2020-12-14:
I tried to use "Install Recommended Firmware" in RealSense Viewer, but firmware update was failed.

wodndb, 2020-12-14:
I get error log. I share this log.

### Test Environment
* 1 m official USB 3.0 cable in product box.
* Connect cable to mother board.
* Firmware update using "Install Recommended Firmware" in RealSense Viewer (ver 2.38.0)

### popup window error log
[image]
```
Started Firmware Update process
Backing-up camera flash memory
Backup completed and saved as 'C:\Users\wodnd\AppData\Roaming\837413020293.20201214_175212.bin'
Requesting to switch to recovery mode
Exception caught in FW Update process-flow: Camera not connected!
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}

ERROR: Recovery device did not connect in time!
```

### Console log
[image]
```
[... cut ...]
17:53:12 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:365 - Found 1 RealSense devices (mask 254)
17:53:12 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\environment.cpp:112 - Found 4 unreachable streams, 4900 extrinsics deleted
17:53:12 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\ds5\ds5-active.cpp:56 - Projector capacity is overrided and disabled by FW
Device PID = 0xad3, Capabilities Vector = []
```

MartyG-RealSense, 2020-12-14:
Thanks for the log.   I have not seen the error **Exception caught in FW Update process-flow: Camera not connected!** before.

Going back to rs-fw-update, if you only have one camera attached to the computer then you should not need to provide the serial number:

rs-fw-update.exe -Signed_Image_UVC_5_12_9_0.bin

When you ran this command, have you put the firmware bin file in the same folder as the one where rs-fw-update is located?  On Windows you can find this folder by right-clicking on the RealSense Viewer launch icon on the Windows desktop and selecting the **Open file location** menu option.  Make sure that the firmware bin file is copied into this location before running the rs-fw-update launch instruction above.

wodndb, 2020-12-14:
First, I checked bin file location.

[image]

And I tried to use rs-fw-update like this.

```
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-fw-update.exe -f Signed_Image_UVC_5_12_9_0.bin

updating device:
Name: Intel RealSense D415, serial number: 836112060407, update serial number: 837413020293, firmware version: 05.09.13.00, USB type: 3.2
 14/12 18:09:20,450 ERROR [400] (types.h:313) hr returned: HResult 0x8007001f: "?
```

I got same error.

wodndb, 2020-12-14:
I tried to update firmware using RealSense viewer with administrator mode, I got some different error logs.

```
Started Firmware Update process
Backing-up camera flash memory
Backup completed and saved as 'C:\Users\wodnd\AppData\Roaming\837413020293.20201214_181728.bin'
Requesting to switch to recovery mode
Exception caught in FW Update process-flow: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "지정된 경로를 찾을 수 없습니다."
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Exception caught in FW Update process-flow: Failed to create FW update device, device id: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}

ERROR: Recovery device did not connect in time!
```

```
18:17:18 [Info]  - librealsense version: 2.39.0

[... cut ...]
18:18:28 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:365 - Found 1 RealSense devices (mask 254)
18:18:28 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\environment.cpp:112 - Found 4 unreachable streams, 4900 extrinsics deleted
18:18:29 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\ds5\ds5-active.cpp:56 - Projector capacity is overrided and disabled by FW
Device PID = 0xad3, Capabilities Vector = []
```

wodndb, 2020-12-14:
Should I install specific .Net framework or visual C++ redistributable package?

MartyG-RealSense, 2020-12-14:
You should not need to install a specific .Net framework or visual C++ redistributable package in order to update firmware.   If you are using Windows 10, you should only need to install the SDK using the installer file from the **Assets** file list at the bottom of the SDK **Releases** page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

[image]

wodndb, 2020-12-14:
Thanks to continuous reply : )

I updated RealSense SDK to 2.40.0 but I got same error.

MartyG-RealSense, 2020-12-14:
Let's try a third method of firmware update.   This will only update as far as firmware 5.11.6.200 but any kind of progress would be welcome.

Please visit the page below and download file **DMFT_Version 6.1.160.14.zip**.   Ignore the page's suggestion to download a newer version, as the newer version has problems with the C# and Unity wrappers.

Unzip the downloaded file and right-click on the file "RS400Camera.inf", and select the 'Install' option.  Then click Yes or Next on all of the following dialog boxes.

wodndb, 2020-12-14:
I got install error when I select install option of "RS400Camera.inf" file. I need to modify inf file to avoid install error.

wodndb, 2020-12-14:
I install using inf file in Windows recovery mode (ref: https://www.supportrix.com/kb/how-to-install-unsigned-drivers-in-windows-10/)

However, firmware update was failed...

MartyG-RealSense, 2020-12-14:
Can you open the RealSense Viewer and click the 'Info' option in the side-panel and post an image here please?

This is the info display from my own D415:

[image]

wodndb, 2020-12-14:
[image]

MartyG-RealSense, 2020-12-14:
Checking back through this case, different serial numbers for the same camera seem to be getting reported.  It has been reported first as 836112060407 in rs-fw-update and the Viewer, and now as 833412061112 in the Viewer info.

I checked my own D415 in the Viewer info and with **rs-fw-update -l** and they both had the same serial number.

I will seek advice about your case within Intel.  Thanks for your patience in the meantime.

MartyG-RealSense, 2020-12-14:
What ypu could do in the meantime is try a **Gold Reset** of the camera.  A corrupted **calibration table** inside the camera has been known to cause firmware update disruption, and a Gold Reset fixes the table.

To perform a Gold Reset, please download the **Dynamic Calibrator** tool fro the link below.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool?v=t

Installing the Calibrator also installs a program called **CustomRW** that can be launched from the Command Prompt.  After installation, input the command below into the Command Prompt to launch CustomRW in Gold Reset mode.

**Intel.RealSense.CustomRW -g**

wodndb, 2020-12-14:
@MartyG-RealSense I'm sorry about camera serial number. I have 4 cameras of D415, and I tested these cameras each one by one. So serial number I reported can be diffrent.

MartyG-RealSense, 2020-12-14:
Thanks for the clarification about havibg four D415.   So the firmware update process fails for all four cameras?

wodndb, 2020-12-14:
Yes ; (

MartyG-RealSense, 2020-12-14:
Multiple cameras having the same problem means that the problem is unlikely to be related to the camera hardware, firmware driver, camera drivers or the USB cables.  It would suggest that the problem is somewhere on the computer.

Do you have another computer that you can test the cameras on?

wodndb, 2020-12-14:
Yes I have another laptop and desktop. I'm going to try firmware update
using these computers tomorrow. Thanks.

wodndb, 2020-12-15:
@MartyG-RealSense I can update firmware in another my desktop.

I reset two desktop what I tested to update firmware and these two computer has same environment to develop RealSense Application based Unity 1 month ago. I will check different between two computers and report to this issue.

Thanks.

closed 2020-12-15 by wodndb

MartyG-RealSense, 2020-12-15:
Thanks very much @wodndb for the update!

wodndb, 2021-01-18:
I can't find what is cause of firmware update fail.
So I restore windows 10.
And I success to update firmware.

MartyG-RealSense, 2021-01-18:
Excellent news that you were able to update firmware after Windows restoration!  Thanks for updating us about your success. :)