realsenseai/librealsense #7988: camera-imu calibration by Kalibr. OverflowError: bad numeric conversion: positive overflow

TouchDeeper, 2020-12-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-128-generic             | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0                      | 
| Language                            | C                         | 
| Segment			|  Robot                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi,

I'm sorry to ask about a problem with Kalibr here. I have posted an [issue](https://github.com/ethz-asl/kalibr/issues/401) in the Kalibr repository about my problem. I want to calibrate the extrinsic between the camera and IMU. But it seems like the timestamp of IMU causes the `kalibr_calibrate_imu_camera` of Kalibr crash. How do you calibrate the extrinsic between the camera and IMU? Could you please provide some advice? Thanks a lot!

MartyG-RealSense, 2020-12-14:
The link below has a project that provides  an IMU calibration file for D435i.

https://github.com/SainaRez/navigation_scripts#calibration
https://github.com/SainaRez/navigation_scripts/tree/master/Utility/imu_intrinsics

TouchDeeper, 2020-12-14:
Hi @MartyG-RealSense 
Thank you for your reply, but it seems like he did not have the same problem as me.

MartyG-RealSense, 2020-12-14:
Their calibration command follows the same structure as the one that you posted in your Kalibr topic.  Does the D435i yaml file provided in that project not assist with your own calibration problem please?

closed 2020-12-16 by TouchDeeper