realsenseai/librealsense #7989: Failed to resolve the request (C# wrapper)

LucaFranceschini, 2020-12-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Windows 10 (version 1507) | 
| Platform                                 | PC  |
| SDK Version                            |  2.0.50727                          | 
| Language                            |  C#                          | 
| Segment			|  ?                         | 

### Issue Description
I'm using the C# wrapper for color image capture, and I tried to follow the example with the following result:

```c#
var configuration = new Config();
configuration.EnableStream(Stream.Color, Format.Rgb8);
var pipeline = new Pipeline();
pipeline.Start(configuration);
```

The code above fails with the following message:

	Camera pipeline failure: 
	Failed to resolve the request: 
		Format: RGB8, width: 1280, height: 720
	
	Into:
		Formats: 
		 YUYV

The example programs based on the C# wrappers work correctly.

The same code, with the same camera, actually works on another machine with the same OS and the same driver. I can't seem to find any difference between the two. I saw other issues mentioning USB 3, but according to the RealSense viewer the camera is correctly recognized as USB 3.

What am I doing wrong? Can I make this more robust? Any idea about possible differences between the two machines?

MartyG-RealSense, 2020-12-14:
Hi @LucaFranceschini   As a starting point in investigating your case, may I ask please whether the RealSense UWP driver version **6.11.160.21** might have been installed on the computer that experiences the problem.   This version of the UWP driver has been reported by a small nuber of RealSense users to cause errors with the C# and Unity wrappers.

LucaFranceschini, 2020-12-14:
> Hi @LucaFranceschini As a starting point in investigating your case, may I ask please whether the RealSense UWP driver version **6.11.160.21** might have been installed on the computer that experiences the problem. This version of the UWP driver has been reported by a small number of RealSense users to cause errors with the C# and Unity wrappers.
> 
> https://downloadcenter.intel.com/download/29827/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

Thanks for your prompt reply. I don't remember doing that, but just in case, how can I check? From the device manager all I see is the Microsoft driver from 2006.

MartyG-RealSense, 2020-12-14:
Could you do a search on your computer please for **rs400camera**

[image]

LucaFranceschini, 2020-12-14:
> Could you do a search on your computer drives please for **rs400camera**
> 
> [image]

Done, no results.

MartyG-RealSense, 2020-12-14:
**Failed to resolve the request** typically occurs when the SDK either does not understand a stream definition (e.g because of a mistake in the code) or because the stream configuration requested is not currently supported by that particular camera.

I cannot find any problems in your code.  So that does make me wonder if the C# program is detecting the connection as USB2, even though it appears as USB3 in the Viewer.

A way to test this possibility would be to use a color configuration that would be supported on USB2 and see if the program runs on the affected computer.  The number in the end of the bracket defines the FPS (30).

 configuration.EnableStream(Stream.Color, 424, 240, Format.Rgb8, 30);

LucaFranceschini, 2020-12-14:
> **Failed to resolve the request** typically occurs when the SDK either does not understand a stream definition (e.g because of a mistake in the code) or because the stream configuration requested is [...]

This resulted in the same error (except of course width and height).

MartyG-RealSense, 2020-12-14:
If the problem reported by the error is with color conversion to YUYV, could you try replacin the color stream with a depth stream, please?   If you get the same error, this would suggest that it is an issue other than the stream definition (as suggested by **Camera pipeline failure**).

**configuration.EnableStream(Stream.Depth, 640, 480);**

LucaFranceschini, 2020-12-15:
> If the problem reported by the error is with color conversion to YUYV, could you try replacing the color stream with a depth stream, please? If you get the same error, this would suggest that it is an issue other than the stream definition (as suggested by **Camera pipeline failure**).
> 
> **configuration.EnableStream(Stream.Depth, 640, 480);**

Oh, sorry, the "Camera pipeline failure" bit was part of my own log message, not the actual error message. Forgot to remove it.

I'll follow your suggestion and report back.

LucaFranceschini, 2020-12-17:
... meanwhile, the computer showing the problem went through a format and new Windows installation, and it now seems to work. It would still be useful to know what could cause this problem to avoid it in future, but feel free to close the issue. Thanks for your help.

MartyG-RealSense, 2020-12-17:
Sometimes problems do get resolved immediately just by performing a fresh installation of the operating system, and you never find out what the exact cause of the problem was.  Sometimes an installation's files get in a tangle and a clean re-install of the OS removes that tangle.  Bearing that in mind, I will close the case.   Thanks!

closed 2020-12-17 by MartyG-RealSense

LucaFranceschini, 2020-12-18:
@MartyG-RealSense I'm tagging you in case you don't get notified from closed issues, sorry if it's not needed. I think I know what the problem was: another process was using the camera.
By reproducing the error, I consistently get the error message in the OP when multiple processes try to access the camera concurrently; it is expected that this do not work, but the error is somewhat misleading, suggesting the requested format is the problem. Don't know if this is a known issue, but I thought it could be useful for others.

MartyG-RealSense, 2020-12-18:
Thanks very much for the update @LucaFranceschini   :)

There are indeed rules that govern how cameras can be accessed by multiple cameras or multiple processes.  Once a stream type ('endpoint') on a particular camera is claimed then the same stream type on that particular camera cannot be claimed by another process until the claim is released.

https://dev.intelrealsense.com/docs/software-support-model#section-multi-streaming-model