realsenseai/librealsense #7990: Invalid Value in rs2_open_multiple, Failed to resolve the request:

HuangWeiChu, 2020-12-14:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|ffffffffffffffff|
|**Firmware Version**|01.05.00.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|ffffffffffffffff|
|**Firmware Update Id**|ffffffffffffffff|

Invalid Value in rs2_open_multiple(sensor:0000023263FFB490, profiles:0000023271462FA8, count:2):

Failed to resolve the request: 
	Format: Z16, width: 640, height: 480
	Format: Y8, width: 640, height: 480

Into:
	Formats: 
	 Y8
	 Z16

RealSenseSupport, 2020-12-14:
Hi @HuangWeiChu 

Thanks for the information. We'd like to discuss with you directly about the issue you're seeing. We have some questions.

Can you submit a ticket at the following location? http://realsensesupport.intel.com/

HuangWeiChu, 2020-12-15:
@RealSenseSupport 
Did you mean the receipt?
I bought this 3 months ago, and I already send an invoice with the receipt for the payment from my school.
So...

Is there anything else I can do?
Cause I've been bothered by this problem twice.
Hoping you can help me deal with this problem.

ctheodorak, 2021-01-26:
Same problem here....

RealSenseSupport, 2021-01-27:
Hi @ctheodorak 

Are you saying you have the same issue where your L515 camera is showing this "fffffffffff" in the serial number section?

closed 2021-02-16 by RealSenseSupport