realsenseai/librealsense #7994: Cubemos no longer supports IntelRealSense

VisionaryMind, 2020-12-15:
Yes, this is not the place to talk about Cubemos, if not for the fact that 1) Intel lists the SDK as "our Skeletal Tracking SDK" on the main product page and 2) Cubemos does not maintain their SDK. So this *is* an issue for both companies but, more important, for current and future Intel users.

Cubemos is defunct, as a company, in general. One look at their website should tell you that. Seeing Intel consistently refuse to associate themselves with their efforts on a technical level (but still promote their product as if it is their own) should tell you much about the growing paucity of ethics present in a once very diligent, politically conscientious company.

That being said, Cubemos code is full of holes and they apparently don't update their software on a consistent basis. Visual Studio solutions are geared toward version 2017, and Unity must be set to 2018.x Their own site refers back to Intel RealSense for "more information about the skeleton SDK", and here in this Github forum -- which is in fact the only place in the whole world that so much as mentions Cubemos, who lacks any social media presence at all) --- we have an Intel RealSense rep sending people back to Cubemos when they know Cubemos will send them back to Intel in an endless circle. It's a ridiculous situation, really. Intel should know Cubemos does not respond to pre-sales inquiries, and I personally have doubts they would respond post-sale either. At least from our experience.

In reality, Intel shouldn't be promoting a product they don't fundamentally understand or support nor should Cubemos be claiming that their software / API is 100% compatible with RealSense hardware or its SDK. It most certainly is not. FYI, this SDK only tenuously works with RealSense hardware and that's only for the pre-compiled CSharp executables. But even if you do get it working, the data you will obtain is not going to be useful, mostly because of the enormous amount of noise coming out of the D-series depth cameras. It creates jitter and displacement of skeletal joint points per frame. It get's lost completely if limbs cross. To make matters worse, it looks earily famiiliar (if not idential) to the OpenCV skeleton recognition module. Take that as you will.

To make matters worse, their SDK is also unable to make use of GPU computing, even though it is clearly listed as an option in the GUI settings. Try turning it on, and Cubemos will tell you "you're hardware isn't robust enough for this feature". We have both an RTX 2080 and Tesla GPU, so it is clear they have not structured support around GPUs. That makes this SDK categorically "non-edge". So the question arises -- why an SDK in the first place and why, in particular, this SDK? But more importantly --- why is Intel willing to promote it but completely unwilling to support Intel users that want to use it? You are engaging in double-mindedness and this is frustrating for customers.

The only reason I am commenting here is that we wanted to give the D455 depth camera one more try (after it failed comprehensively on all tests for our 3D scanning pipeline). Our representative in Sweden sat on a call with a certain Merritt Hope, and eventually, Intel came back with the statement that "their camera was really not geared toward high-end commercial uses and is, more or less just for hobbyists". That's all fine and well. But here we have downloaded the Cubemos Skeleton SDK with the hope that it would allow us to capture motion and translate that to yet another CG pipeline we own and ---I will be brief:

1) Cubemos Python bindings for Skeleton SDK and RealSense don't work on the latest Windows 10 build. It's own DLL cannot be interpreted by CPP ground zero builds either.

2) The Unity package is also malformed and not properly developed. It apparently refers to the wrong versions of DLLs (or something to that effect) and even dropping realsense2.dll and Intel.Realsense (built fresh from librealsense2 source) into the Unity plugin directory --- Unity is unable to load them. I am sure RealSense users experiencing these issues eventually reached out to Cubemos, they would come to our mutual conclusion that Cubemos is not able or willing to support its own product.

If we sound exasperated, it is because we are. We have wasted countless hours on inefficient, sloppy code from a company that Intel should know better than to back. Cubemos only has short of 3,000 customers and if they keep up at the same pace, they certainly won't retain them. 

Intel continues to step on its own feet, sending its VP of engineering out into the field to query "disgruntled users" when, in fact, the problem is not with the users' perception but with the lack of vision for the product / software line, in general. That is why companies like DotProduct are able to make a killing with their software. It fills gaps that shouldn't be there in the first place.

RealSenseSupport, 2020-12-17:
@VisionaryMind Thank you very much for bringing up this issue. We will reach out to Cubemos and investigate.

RealSenseSupport, 2020-12-22:
Hi @VisionaryMind Cubemos has a beta version where the samples work with the D455, it will be launched soon.

seashairo, 2021-01-18:
When is soon in this context? I'm seeing very similar issues and there's a huge lack of support. 

While some of their samples technically do work with a D455, only the precompiled binaries work. Self-compiling them does not, and the instructions are incomplete. 

Even the ones that do work do not work with GPUs. I was even told by their support that the SDK has problems running on machines with an external GPU and to *disable my GPU* to continue working with it. Apparently "No dedicated GPU needed" means "No dedicated GPU supported". Honestly the worst part about that is that it doesn't support external GPUs in a game engine.

I know this comes off as a bit of a rant, but I feel like choosing to try Intel's advertised skeletal tracking has been the worst decision I've made for a new project.

viveksavsaiya, 2021-01-21:
I do have a bitter experience with Cubemos, We bought two license. and this is the replay from support team.

**I’m sorry you continue to experience Problems with the Unity Plugin. Unfortunately we are unable to provide unity-specific support until at least 01.02.2021.
If this is a deal-breaker for you, I would advise you to engage the intel shop (where you bought the license) for a refund.** 

I'm really shocked.

RealSenseSupport, 2021-02-08:
We discussed with Cubemos on the Unity support. Since the issues for the Unity support, we have removed all mentions of Unity our website. Sorry for the inconvenience.

doukasd, 2021-02-24:
Thank you for bringing all the issues up @VisionaryMind and @seashairo, I just installed their trial "SDK" with the intent of using in Unity but quickly realised this doesn't look like what I would expect and thankfully your post came up quickly before I spent too much time on this. Now off to try the Nuitrack SDK and see if it's any better... Unless you have any other alternative to recommend for skeleton tracking in Unity using RealSense cameras?

(PS: Azure Kinect skeleton tracking in Unity works great and I just bought the RealSense D435i to compare quality)

rnrytjr2, 2021-03-05:
I hope it can help you by activating IGPU in the BIOS.

jasjang, 2021-04-07:
I was very disappointment with the Cubemos SDK,
When I use the newest 3.1 version, and import the newest unity package, still show the error to load external model.
When I use the 2.2.1 which provided from Cubemos to temporarily solve the unity problem, the D455 can't be recognized via 2.2.1 version.

After more than half year, the unity model error still... And I was struggled!
my client use the D455 which 2.2.1 incompatible...

I think Intel should pay attention to this problem when the site still sell the package.

Dhelio, 2021-05-05:
Thank you @VisionaryMind for the thorough overview of the same problems that I've had for about a day, and thank you @seashairo  for the follow up and the alternative.

Frankly it's kind of frustrating, trying to setup some SDK, only to find that it's mantained improperly, to put it lightly.

wlghks0723, 2021-05-28:
Compared to other sdks and other programs provided by intel, there is too little data or explanations through GitHub.

Finding materials such as forums or bulletin boards to solve certain errors It is very difficult because there is no environment for sharing with each other.

I recently saw the sdk updated on the intelrealsense site and started using cubemos again, and such notices were only available through twitter or sns, not through intel sites or other methods.

In addition, in the case of other skelton sdks, whether there is a difference between the free version and the purchased version, the tracking system is based on the cloud and edge, etc., it is not easy to know the data, and it is not easy to inquire where there is a difference.

For the support provided by Intel, we are using the free version with intelrealsense purchase status to see if the cubemos sdk is suitable for the current project. Because there is no way to get help.

You may have misunderstood the wording on the site, but I thought that the inquiry part after purchase was supported.

Because of this, I can't find a solution to all the issues I face in the process of checking if this SDK is right for me before buying.

ioannismihailidis, 2021-06-21:
Shouldn't it be easy to switch from cubemos's to any other skeleton detection sdk? There are a lot of deep learning based implementation out there.

The cpp realsense/skeleton tracking example shows that the cudemos is not deep integrated in the pointcloud/depth algorithms of librealsense. In fact cubdemo is using just the rgb image to detect joint positions in 2d rgb space. In the next step they retreive the 3d positions from the depth image (rectified to 2d rgb space).

Vangos, 2021-07-20:
In case you are still looking for alternatives, you could check the [LightBuzz SDK for businesses](https://lightbuzz.com). It's using the Color + Depth feeds to produce truly 3D views and supports the latest version of the RealSense SDK + Unity3D.

GrubbyHalo, 2021-07-27:
I downloaded the trial version of the cubemos skeletal SDK today with high hopes. I have spent almost an entire day on trying to get it up and running. The dependencies  for Linux are outdated. The required OS for Linux is Ubuntu 18.04. We are now on 21.04 and you can't get libboost 1.65 anymore. Its incredibly frustrating. I can't understand how a company tries to sell and market its product but clearly don't maintain it.

wlghks0723, 2021-07-28:
I don't know what project you're working on
If the environment is not
Try nuitrack
That service is much better managed.

As a result of using both, cubemos seems to have a better recognition rate, but there is no noticeable difference. However, in the case of CUBEMOS, there is an advantage that AI-applied models can be used as a trial version. But if you're just tracking or testing and you're running into environmental issues, try NUITRACK.

wlghks0723, 2021-07-28:
> 오늘 큰 기대를 하고 계시는 인디아나레탈 SDK 체험판을 받았습니다. 날짜와 시간을 운영하고 있습니다. Linux에 대한 최적의 구식입니다. Linux에 필요한 OS는 Ubuntu 18.04입니다. 우리는 이제 21.04에 더 이상 libboost 1.65를 수 없습니다. 사용하고 있습니다. 제품이 판매되고 있으며, 이에 따라 매출이 증가하고 있습니다.

I don't know what project you're working on
If the environment is not
Try nuitrack
That service is much better managed.

As a result of using both, cubemos seems to have a better recognition rate, but there is no noticeable difference. However, in the case of CUBEMOS, there is an advantage that AI-applied models can be used as a trial version. But if you're just tracking or testing and you're running into environmental issues, try NUITRACK.

mariasni, 2021-08-14:
Hi, I have similar problems to use cubemos as python wrapper on windows. Tried reinstalling multiple time and keep getting the errors as below: 
[image]

[image]

There is little feedback by cubemos support email, sadly. If you encountered similar issues and managed to fix them- I would really appreciate a piece of advice!

RealSenseSupport, 2021-08-20:
Let's move the conversation to Cubemos forum. Thanks!