realsenseai/librealsense #7995: realsense-viewer problems for d455

Ctiger96, 2020-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400(d455)  | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Linux(Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-58-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2.40.0                     | 
| Language                            |  C++                        | 
| Segment			|  Robot                          | 

### Issue Description
I installed SDK using this method: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
When I run `realsense-viewer` in the terminal, three Warnings and one Error appear:
```
15/12 13:32:24,547 WARNING [140438733784832] (backend-v4l2.cpp:1271) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
15/12 13:32:24,582 WARNING [140438733784832] (sensor.cpp:1076) RGB Camera: skipping Exposure control. descriptor: [min/max/step/default]= [0/0/0/0]
15/12 13:32:24,584 ERROR [140438733784832] (types.h:313) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
15/12 13:32:24,584 WARNING [140438733784832] (sensor.cpp:1100) Failed to add Enable Auto Exposure control for RGB Camera

```
Would you plz tell me what to do with it? Thx!

One the other hand, PointCloud is jittering seriously, and is it normal? How can I modify it? Thx!

[image]

Jesburg, 2020-12-15:
I have the same problem, seeking for the way to deal with it!

Ctiger96, 2020-12-15:
@MartyG-RealSense  Hello sir, would you please give me some advice? Thank you very much.

RealSenseSupport, 2020-12-15:
The warning messages for RGB camera doesn't appear to show up on my system with the same Ubuntu and kernel version. (See below )
```
:~$ uname -r
5.4.0-58-generic
:~$ dpkg -l | grep "realsense"
ii  librealsense2:amd64                              2.40.0-0~realsense0.3814                            amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-dkms                               1.3.14-0ubuntu1                                     all          Modified kernel modules for librealsense2
ii  librealsense2-gl:amd64                           2.40.0-0~realsense0.3814                            amd64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-net:amd64                          2.40.0-0~realsense0.3814                            amd64        Intel(R) RealSense(tm) - Network Bridge extension
ii  librealsense2-udev-rules:amd64                   2.40.0-0~realsense0.3814                            amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  librealsense2-utils:amd64                        2.40.0-0~realsense0.3814                            amd64        Intel(R) RealSense(tm) Camera Capture API - utils and demos

```
[image]

Please try to recalibrate the camera as I suspect the jittering might be caused by the camera calibration is slightly off. 
https://github.com/IntelRealSense/librealsense/issues/7615#issuecomment-712767503

Ctiger96, 2020-12-16:
@RealSenseSupport The information you showed me is totally the same as mine. And now I uninstall librealsense using `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` and want to reinstall it using `sudo apt-get install librealsense2-dkms`, then an ERROR appers:
```
Loading new librealsense2-dkms-1.3.14 DKMS files...
Building for 5.4.0-58-generic
Building initial module for 5.4.0-58-generic
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.14: added
nvidia, 440.100, 4.15.0-128-generic, x86_64: installed
nvidia, 440.100, 5.4.0-53-generic, x86_64: installed
nvidia, 440.100, 5.4.0-58-generic, x86_64: installed
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d 	
```
I dont know how to deal with the `dkms.conf` error, and I think this is the reason why I got the WARNING when run the command `realsense-viewer`(at the first installing time, I didn't watch the information on the terminal carefully). Would you plz give me some advice? Thx!

Ctiger96, 2020-12-16:
@RealSenseSupport I tried to re-calibrate the camera, but I got the worse result than I showed. So how can I recover the default parameters of the camera?

RealSenseSupport, 2020-12-16:
Yes, the patch can also be manually built using the steps [1-4] in "Prepare Linux Backend and the Dev. Environment":
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Please use CustomRW tool with -g option to reset the calibration table to it's  factory setting.
[image]

RealSenseSupport, 2020-12-23:
Just wondering if any further help is needed for this issue. Thank you.

closed 2020-12-29 by RealSenseSupport