realsenseai/librealsense #7998: How to set DBUILD_NETWORK_DEVICE flag to open in Linux?

ranjitkathiriya, 2020-12-15:
To enable RealSense device over a network, please build the SDK with CMake flag -DBUILD_NETWORK_DEVICE=ON.
This binary distribution was built with network features disabled.

MartyG-RealSense, 2020-12-15:
Hi @ranjitkathiriya   The open source ethernet networking white paper states that "No network code is compiled into core realsense2 module, limiting code size and security risks for customers who don’t need this functionality".

ranjitkathiriya, 2020-12-15:
Actually, I am using Lips AE400 camera, so for that, I need this flag to set ON

MartyG-RealSense, 2020-12-15:
RealSense-compatible third-party cameras sby non-Intel manufacturers are not provided with technical support from Intel as they are not Intel products and the manufacturer of a particular third-party camera is instead responsible for supporting it.  LIPS Corp would therefore be the appropriate people to discuss this with.

ranjitkathiriya, 2020-12-15:
Thanks for your help and for making this thing clear.

closed 2020-12-15 by ranjitkathiriya