realsenseai/librealsense #8001: Question on Realsense camera and projector calibration

ogy96, 2020-12-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.39.0 }                          | 
| Language                            |  {C#/EmguCv }                          | 
| Segment			| AR}                          | 

### Issue Description
I am currently working on an AR Sandbox using a Realsense D415 and a projector.

How can I calibrate the Realsense camera so that it matches the projector's image?
Are there any projects that calibrated a D400 series camera with an external projector?

Also is there a way that I can access the coordinates of the colour and depth image so that I can see if they match and if thy are automatically calibrated?

MartyG-RealSense, 2020-12-15:
Hi @ogy96  In general, an external projector that generates a dot-pattern projection does not need to be calibrated to a 400 Series camera, as it is not required to be co-located with the depth sensor.  You can move the external projector around and shake it.  Intel has published a white-paper document about projectors. 

https://dev.intelrealsense.com/docs/projectors

It sounds as though you may be referencing projecting an image from the projector and interacting with the scene with a 400 Series camera.  A company called LUMOplay set up a system similar to this with an external projector and a RealSense L515 lidar camera.  this project may provide some useful insights for an equivalent with D415.

https://www.youtube.com/watch?v=J8W1gLi3Idk

In regard to retrieving coordinates: if you are using C#, the information in the link below may be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/7524#issuecomment-705517349

ogy96, 2020-12-15:
Hi @MartyG-RealSense yes exactly the projector is supposed to project the topography image on the sand. And while playing with the sand the depth camera is supposed to change the topography image according to its depth.
Therefore the Realsense Camera and the projector images must be calibrated.

I thought of something like this: https://bingyaohuang.github.io/Calibrate-Kinect-and-projector/
I was wondering if someone has tried something similar with a Realsense instead of a Kinect and EmguCv instead of OpenCv

Thanks for the info on retrieving coordinates

MartyG-RealSense, 2020-12-15:
I would imagine that if the depth camera and the projector were mounted beside each other and the display output from the computing the device the camera is attached to was sent to the projector, then you would have the foundations of a solution.  

1.  The camera depth image emitted by the projector falls onto the three-dimensional surface of the real-life sand and bends to the contours of the sand.   

2.  The depth camera meanwhile can read the current depths of all the pixels on the digital depth image of the sandbox and do work on that data, such as assigning color-grading to the data based on its distance from the camera.   As users move the real-life sand, like in Tanked, the depth measurements should update in real-time.

ogy96, 2020-12-15:
@MartyG-RealSense  thanks for the advice! I will start working on it

closed 2020-12-15 by ogy96