realsenseai/librealsense #8004: Error setting exposure ROI on color camera: hwmon command failed (response -6= Invalid parameter)

ttencate, 2020-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Arch Linux  | 
| Kernel Version (Linux Only)    |  5.9.10-arch1-1   | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.0 | 
| Language                            |  C++  | 
| Segment			|  n/a  | 

### Issue Description

I am unable to set the auto-exposure ROI on the color (RGB) sensor; the `set_region_of_interest` function throws `rs2::invalid_value_error` (see output below).

Here's a minimal example:

```c++
#include <librealsense2/rs.hpp>

#include <iostream>

int main() {
  rs2::config config;
  config.enable_stream(RS2_STREAM_COLOR);
  rs2::pipeline pipeline;
  rs2::pipeline_profile profile = pipeline.start(config);
  rs2::roi_sensor color_sensor = profile.get_device().first<rs2::color_sensor>().as<rs2::roi_sensor>();

  // Getting ROI works fine.
  rs2::region_of_interest roi = color_sensor.get_region_of_interest();
  std::cerr << roi.min_x << ',' << roi.min_y << ' ' << roi.max_x << ',' << roi.max_y << '\n';

  // Setting ROI throws an exception.
  color_sensor.set_region_of_interest(roi);
}
```

The output:

```
160,120 480,360
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x75( 78 168 a0 1e0 ) failed (response -6= Invalid parameter)
Aborted (core dumped)
```

The above is linked with librealsense 2.38.0 that I compiled myself. I did not set `FORCE_RSUSB_BACKEND=ON` when building, so I think I would have gotten the V4L2 backend.

It works fine in the RealSense Viewer, although that's a 2.39.0 build.

Similar errors were reported in:

* #6662 (closed for lack of customer response)
* #5308 (closed by customer)

MartyG-RealSense, 2020-12-15:
Hi @ttencate   You may find the C++ script in the link below helpful.  It describes how to define both a depth and RGB ROI and shows how to catch the **invalid value** error when it is thrown.

https://github.com/IntelRealSense/librealsense/issues/4015#issuecomment-595205511

ttencate, 2020-12-15:
Thank you. I can catch the exception, but the fact that there is an exception in the first place looks like a bug in librealsense to me.

MartyG-RealSense, 2020-12-16:
Catching an error from a failed attempt to apply a setting and retrying in a loop until an instruction succeeds is a valid way to handle such a problem, as long as it does not take too long (e.g more than 5 tries) to succeed.

ttencate, 2020-12-16:
Thank you, that is what I was doing already, so I'll just continue doing that.

However, coming from a software background, I have to disagree that this is a "good" solution. Doing the same thing twice under the same circumstances should yield the same results. Software that does something different under the same circumstances violates the [principle of least surprise](https://en.wikipedia.org/wiki/Principle_of_least_astonishment).

I appreciate that hardware and timing can be finicky and that retrying is sometimes necessary as a last resort, but an abstraction like librealsense exists to shield the user from such implementation details as much as possible. If for some reason the issue can't be fixed in the library itself, at least a warning could be added to the [`set_region_of_interest` documentation](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1roi__sensor.html#ab43cb7bc5be9c83fca96d26be1bfdf78) that the function might randomly fail, and that the user needs to add a retry loop around it.

Still, because it only happens for the color sensor and not the depth sensor, I get the feeling there's something deeper going on that could potentially be fixed without even needing a retry loop.

MartyG-RealSense, 2020-12-16:
There is a lengthy case that also involves problems with setting the ROI.  I will link you to advice provided about setting ROI from a RealSense team member.

https://github.com/IntelRealSense/librealsense/issues/6952#issuecomment-665254844

MartyG-RealSense, 2020-12-23:
Hi @ttencate  Do you require further assistance with this case, please?   Thanks!

ttencate, 2020-12-23:
It's not a support case, it's a bug report :) But if you insist that undocumented random failures are expected behaviour, feel free to close it.

MartyG-RealSense, 2020-12-23:
As you have mentioned in https://github.com/IntelRealSense/librealsense/issues/8004#issuecomment-746525472 that you have a workable solution, though not one that is entirely satisfactory for you, I will close.

closed 2020-12-23 by MartyG-RealSense