realsenseai/librealsense #8007: Get the number of points from a depth frame

louis-etne, 2020-12-15:
| Required Info                         |                                               |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |  Linux  |  
| Language                            |  C++  | 

### Issue Description
I need to get the number of points from a depth frame after being processed by each filters. So for the moment I use this:
```c++
rs_points = rs_pointcloud.calculate(f);
output << rs_points.size() << ";";
```
But it is extremely slow, without I can process approx 40 frames per second and with it I drop to only 13 frames per second. My question is, is there a way to get the number of points from a depth frame directly ?

Thanks for your help.

MartyG-RealSense, 2020-12-15:
Hi @louis-etne   Do you have different results if you use **std::cout** instead of **display**?

**std::cout << rs_points.size() << ";";**

Would it also be practical to put the display code within a keypress check so that it only updates the value when a particular key is pressed?  For example, the C++ script in the link below has a mechanism for checking if 'q' is pressed and take an action if the key is pressed.

https://github.com/IntelRealSense/librealsense/issues/6337

[image]

louis-etne, 2020-12-15:
Thanks for your answer.

Unfortunately, my code doesn't take any user input, it runs by itself on a headless raspberry pi. I need to get the number of points from the depth frame within an infinite loop where all the processing appends and write the result to a `csv` file to analyse the timing after. And i need to get the number of points after each filter, so I need it to be quick.

MartyG-RealSense, 2020-12-15:
Writing to csv in librealsense is slow.  The most efficient recording method is a **bag** file, though it requires a large storage drive because the file sizes can be tens of gigabytes for a prolonged recording duration..

https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380728118

closed 2020-12-30 by MartyG-RealSense