realsenseai/librealsense #8008: Analyzing post processed depth data.

arothenberg, 2020-12-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I am using post processing to get a less noisy depth image via rs-post-processing.cpp in the sdk . That part works. I want to then use the new post processed depth table for analyses (ex find the highest point within a narrow distance/de[th range) .

 I assume using your library is probably better than making my own filters to clean noise. I notice that  in the sdk examples a point cloud is used to handle the post processed table , I assume a point cloud is used  because there are libraries to change angle and perspective with point clouds.  I don't need anything but the depth data the camera renders. I don't need perspective or fancy rendering. 

I also want to use the remove background code in rs-align-advanced but there is also a  threshold variable in post processing I could use instead. 

2 questions since I'm having some difficulty fusing the code between the two examples. 
1.  Is it reasonable to use the depth table after post processing for analyses or is this wrong? I've had trouble today with this approach?
2. do I want to use the remove background in rs-align-advanced OR should I just set the threshold ranges in post processing?

speed is important. I want the fastest method. 

Thanks.

MartyG-RealSense, 2020-12-16:
Hi @arothenberg   I considered your needs carefully.  The most straightforward way to achieve better speed may be to simply be selective in the types of post-processing filter that you choose to apply.  For example, more than one type of filter can provide hole-filling, and using a **spatial filter** may be slow, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/4468

In regard to your questions:

1.  If you are aiming to use post-processing primarily to reduce noise then there may be other methods that you can use to improve results instead of applying post-processing (which is performed on the computer's CPU and not the camera hardware).  Intel's camera tuning guide is an excellent resource for image improvement.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

Examples of noise reduction methods could include not using the camera's dot-pattern projection (the 400 Series cameras can alternatively use ambient light to depth-analyze a scene), and avoiding using the camera near fluorescent ceiling lights if possible (or matching the camera's FPS closer to the lights' operating frequency if you cannot avoid such lights - this is 30 FPS for some lights and 60 FPS for others).

2.  The depth to color alignment required to subtract the background in **rs-align-advanced** would have a processing cost (the enabled color stream, and the alignment operation).  And if you do not feel that you would benefit from using a color stream in addition to the depth stream (for example, to better distinguish the background pixels from the foreground pixels), then by all means use the **threshold** filter on depth data only.

arothenberg, 2020-12-16:
Thanks. The last link(spatial filter hole filling) looks promising but the first link says that spatial filter might be too slow. I'll play around with it anyway and see what happens.  

Yeah, I don't need color data at all so I'll just use the threshold variable. 

Should I still use the remove background example as a model for building my own filters? For example I may make a a simple noise filter that overwrites extreme depth values that are only one  pixel in length - maybe  I'll overwrite the bad pixel with a good adjacent pixel.  I assume using the memset function as in remove_background is the optimal method(?). 

I'll also recalibrate the camera and try your suggestions.  Thanks again.

MartyG-RealSense, 2020-12-16:
As the RealSense SDK is open source, you are free to create your own adaptations to meet your project needs.  Before doing so though, the discussion below may be a useful resource for identifying and dealing with invalid values.

https://community.intel.com/t5/Items-with-no-label/filtering-out-invalid-depth-values/td-p/676084

arothenberg, 2020-12-16:
Looks like a good link. Thanks.

arothenberg, 2020-12-17:
So I re-opened this because I have a quick question that the previous post led to.

I played around with in-camera settings and now I realize that I never even needed to code any post processing because your portal has those options and  I only  need one set up.   So I tuned the camera settings to something that looks pretty good for my purposes. 

It's not noisy but there is a bit of noise that I'll have to filter.  Both "hole filling filters" (post process) clean up a lot of the noise that's left. 

Is your "hole filling" filter fast for most cases ie would I gain anything  by  writing my own simple "hole filling" filter - traversing the depth table and replacing bad depths with adjacent good depths? 
Thanks.

MartyG-RealSense, 2020-12-17:
I would refer you to Intel's own post-processing guide to decide whether or not to write your own filter or use Intel's.  If you do a browser Find operation on the page for the word **hole** then it will highlight all mentions of hole-filling for each filter covered, in the guide making it easier to compare the characteristics of each filter that offers hole-filling.

https://dev.intelrealsense.com/docs/post-processing-filters

arothenberg, 2020-12-18:
Thank you. There was some good information in that article. 
It doesn't talk about performance.  But it's description of "Hole filling" filter  is the same filter that I was going to write(above) and I'm sure the developer knows best practices better than me so I'll go with that filter for now. 
Thanks.

closed 2020-12-18 by arothenberg

MartyG-RealSense, 2020-12-18:
Thanks very much - good luck!